iNavFlight/inav #4748: INAV Board and Sensor Alignment

mickey2011, 2019-05-27:
INAV Configurator 2.1.4
Target OMNIBUS F3 and F4
INAV 2.1   25 Feb 2019
INAV preset Z84 Wing

FC board(s) CALIBRATED outside of airframe, all normal.
FC(s) installed into airframe with USB facing toward nose of aircraft, i.e rotated 270 degrees (default USB position is to the right)
INAV Configuration TAB "Board and Sensor Alignment YAW therefore set to 270 degrees.
Back on SETUP TAB FC pitches and rolls correctly, but PITCH is slightly down -5 degrees due to mounting in airframe not being parallel to anticated level flight attitude.
Now to correct this I have gone to Configuration TAB and had to set ROLL to +5 degrees because FC board is rotated through 270 degrees????????
If I set PITCH to +5 degrees ROLL is affected by 5 degrees back in Setup TAB

Is this the intention???????????

Now with the above set up (PITCH and ROLL crossed in Configuration) I setup Adjustments TAB "Board ROLL Alignment to adjust PITCH in flight????

This is what is happening.

The figures that are shown into the GUI fields are those that are inserted into the CLI.

This means the Configuration Board and Alignment settings are affecting the FC differently to the Adjustments setup and is ALL horibly confusing, can someone please look at this and advise.
Thanks

leeph, 2019-05-31:
Roll and pitch do not suddenly become 'crossed'. The roll and pitch angles in the configurator are ALWAYS relative to the forward arrow on the FC. It is not referring to roll and pitch of the aircraft.

The same is true for the adjustments tab. If you have rotated your flight controller 270 degrees (YAW) then yes, from the aircraft's perspective, roll adjustments in the FC translate to a pitch adjustment.

So long as you always think about the roll and pitch from the FLIGHT CONTROLLER'S perspective, you will not get confused.

DzikuVx, 2019-05-31:
yes, alignment of FC is from a FC perspective, not aircraft. This might be counterintuitve at the beginning and was discussed here before

closed 2019-05-31 by DzikuVx

mickey2011, 2019-06-02:
Thanks for the clarification that I asked for.  Entries into the CONFIGURATION tab effect the default board orientation irrespective of an installed board orientaion which in this case was YAW adjusted by 270 degrees. So alter/adjust ROLL to effectively adjust PITCH.

However the second part to my question has not been commented on. This is that adustments via the ADJUSTMENTS tab does automatically take into consideration any YAW adustments in the Board and Sensor alignment. i.e YAW in CONFIGURATION set to 270. In ADJUSTMENTS set up for inflight PITCH adjustment makes adustments to the ROLL value in CONFIGURATION, but adjusts PITCH if board YAW is 0 or 180 degrees, and indeed if YAW is 45 dgrees makes adjustments to both PITCH and ROLL.

Hence my comment about "confusion" in one place 'CONFIGURATION' you have to work this out manually, but in 'ADUSTMENTS' the INAV Gui works it all out what to adust.

Both situations appear to be undocumented.

potater1, 2019-06-09:
What changed in 2.2 from 2.1, I just had a crash when testing 2.2 for the first time, the board orientation was off by 90 degrees using the exact same settings. I had `align_board_yaw` set to 900 when using 2.1, now the correct value is 0. I do have the arrow on the FC pointing to the front, so it's correct now and the previous setting was strange, I thought the clone board just had the wrong arrow direction. OmnibusF4.

mickey2011, 2019-06-09:
Potater1
This needs to be a new ISSUE because this 4748 issue is closed and I don't think there will be any action on what you raise here. I have not tried 2.2 yet but I'm about to and my board will also be at 90 degrees.
After setting the CLI to "align_board_yaw set to 900" did you check back to the SETUP tab to see if your aircraft is responding correctly to Pitch, Roll and Yaw??

Hopefully I will get to 2.2 tomorrow, so why don't you cut and paste your comment above to a 'new ISSUE' and I will also comment to it if I find anything amiss. Cheers

Mickey2011

leeph, 2019-06-09:
You should always test your craft fully on the bench /ground before sending up a new version, let alone a RELEASE CANDIDATE.

Sorry for your crash but that's bad operational practice on your part, not the fault of iNav.