iNavFlight/inav #4750: Tie maximum tilt angle to PID profiles.

Panther2804, 2019-05-27:
Right now, there isn't any way to change your max tilt angle for stabilised flight on the fly.
It would be great if this option could be tied to the PID profiles accessible from the OSD menu.

This would allow quad users to essentially change the maximum speed of their quad on the fly, to make the quad more accessible to beginners (letting your friends fly,...).

teckel12, 2019-05-27:
@Panther2804 Wouldn't ANGLE mode allow this already? You can set a max tilt for ANGLE mode. Then, use HORIZON or ACRO mode for no max tilt angle.

Panther2804, 2019-05-27:
No, I want to set different maximum angles, to switch between "beginner (like 20 degrees)", "intermediate(40)", and "pro(60)" or something like that, I like to fly angle too (for speed tests, landing w/o googles,....), and i have different friends with different skill levels. 
An option in the OSD for tilt in angle mode for all profiles would also work, but would prefer to have this level tied to the pid profiles for easy access.

closed 2019-08-09 by stale[bot]