iNavFlight/inav #4751: FC Matek F722 mini running iNav 2.1.0 froze mid flight

romko1, 2019-05-27:
Hi, I have a MatekF722 Mini, FW: InavFlight v.2.1.0. A model of about 15m from me and the OSD freezing and on the radio commands the model did not respond and a crash followed. Unfortunately, I do not have a flight data record (blackbox), just a short video where OSD freezes in the second second. Previously done about one hundred starts and no problem. CPU load in inav when connected to computer 6%.ESC protocol: Oneshot125.
I apologize if this thing has been dealt with here, but I've never found it. 
It's a critical mistake. Has it happened to anyone?
Please find some solution to the problem? It's dangerous to use it when it's like this.

Video:
https://youtu.be/c5hVB59jTFo

fiam, 2019-05-27:
@romko1 Are you by any chance using an RunCam Split and is it connected to the FC? In any case, please post a config diff so we can see the whole setup.

romko1, 2019-05-27:
@fiam : No, my RunCam Split mini2 is not connected to FC but is powered by VTX from 12V (AKK FX2 ultimate mini: https://www.akktek.com/products/vtx/akk-fx2-ultimate-mini.html)

My config:
# resources

# mixer

mmix reset

mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix
smix reset

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100

# feature
feature -THR_VBAT_COMP
feature -VBAT
feature -TX_PROF_SEL
feature -BAT_PROF_AUTOSWITCH
feature -MOTOR_STOP
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
feature -TRACE
feature THR_VBAT_COMP
feature VBAT
feature TX_PROF_SEL
feature BAT_PROF_AUTOSWITCH
feature MOTOR_STOP
feature SOFTSERIAL
feature GPS
feature TELEMETRY
feature CURRENT_METER
feature PWM_OUTPUT_ENABLE
feature OSD

# beeper
beeper -RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
[... 999 lines cut ...]
C07: FREE 
C08: Wÿ?ËRÚñ?°ø}½hì«¡Ð¿?¯?¿¹óS¸eãhïÐÜm¼5Üçbz1àå?o¥öµn¢#ÎÇ?k`4?¿?%~??ó?[?÷t×?õ?'H0?³oß~g?Qj?üTßSää?MúsÝõöú6ö°OjSIriO^,»/©/\?9UKLo"¢¿?ós©-DZyóO?ÏâÖ½kc1Ù;ãÿ1 OUT
C09: Wÿ?ËRÚñ?°ø}½hì«¡Ð¿?¯?¿¹óS¸eãhïÐÜm¼5Üçbz1àå?o¥öµn¢#ÎÇ?k`4?¿?%~??ó?[?÷t×?õ?'H0?³oß~g?Qj?üTßSää?MúsÝõöú6ö°OjSIriO^,»/©/\?9UKLo"¢¿?ós©-DZyóO?ÏâÖ½kc1Ù;ãÿ2 OUT
C10: SPI3 SCK
C11: SPI3 MISO
C12: SPI3 MOSI
C13: BEEPER OUT
C14: FREE 
C15: MPU CS
D00: FREE 
D01: FREE 
D02: SDCARD CS
D03: FREE 
D04: FREE 
D05: FREE 
D06: FREE 
D07: FREE 
D08: FREE 
D09: FREE 
D10: FREE 
D11: FREE 
D12: FREE 
D13: FREE 
D14: FREE 
D15: FREE

fiam, 2019-05-27:
@romko1 Thanks for the clarifications. There are basically 2 things that could have produced that lock: runcam split (this is fixed in firmware 2.2-rc1) or a faulty/improperly connected i2c device. Given than you have no runcam split connected, that the lock lasted around 10 seconds and that was (incorrectly) the exact i2c timeout for F7, I’m inclined to believe it was the latter.

In the meantime, we’ve reduced this timeout to 10ms (like on F3 and F4), so if this happens It won’t result in a drastic lock.

I’d suggest you review that the compass is properly connected and test it on the bench before flying that machine again.

romko1, 2019-05-27:
@fiam Thank you for your help and explanation. My RunCam split mini2 has FW v.2.3. I have a compass connected to the same I2C as a barometric sensor. The compass is integrated with the Radiolink TS-100 GPS. The compass works fine, but I check it and optionally calibrate it ... there was no interference on that video around the 4m crash.

romko1, 2019-05-29:
Hi, so I made a FW modification: I2C: Reduce timeout on F7 from 10s to 10ms and I also repaired corrupted resource C08 and C09, of course, after installation, calibrated compass, accelerometer. I have also made a short 2-minute flight and yet everything is fine. Thank you for your help and valuable advice.

lunebleue-zz, 2019-06-02:
Hi, I had same behaviour with Matek F722-SE, crashed in mid-flight with INAV 2.1 and then watching OSD video noticed  OSD freezed (looks like processor crashed), First time I crashed I thought it was because of D-SHOT (D-SHOT 300 in my case) so I moved to multishot and same happened. Thought it was my NEW Matek board that was defective so replace it with the Matek F722 STD, Same happened again... So I'm curentlly running BF, Please fix the bug with these F722 boards so that I can go back with INAV  :)

romko1, 2019-06-03:
Hi, So far, the problem has not been repeated, but when connected to the battery, I see in the OSD HDOP 0.0 and about 20s a bad GPS (beeper sand). Subsequently, HDOP is 99 and GPS are working. After approx. minute (cold start) FC reports me a 3D fix. I checked the TX-RX between GPS and FC, also cold contacts. GPS I have Radiolink TS100 mini. It worked correctly before updating FW. It's strange. Can you please tell me where the error is? Thank you.

digitalentity, 2019-06-04:
The freeze itself is an indication of bad wiring. The fix in #4753 allows i2c to recover, but you should consider improving the wiring (maybe adding 2-3K pull-up resistors on SCL/SDA lines).

The GPS failure issue looks like a duplicate of #4411 which will be addressed once I'm able to reproduce (so far all my GPS modules are working fine and I don't have any hardware where I can debug this).

closed 2019-06-04 by digitalentity

romko1, 2019-06-05:
@digitalentity : I found out where the mistake is (#4411). As I wrote above I have a GPS module Radiolink TS100 mini. If I connect to a given GPS module via FTDI cable (u-center) I preset it (transfer rate, frequency ... etc) and save, then connect GPS module to FC MatekSys F722 mini it works fine. However, if the GPS module power is longer off (more than 3 hours) then the GPS module will lose my settings (defaults to 9600 speed, frequency) and will turn on the FC MatekSYS F722 mini - automatic configuration approx. 20 seconds red - buzzer bleeping (HDOP is 0.0 not 99), and then get caught - shining blue (HDOP is 99). Wouldn't it help to set a larger delay in the firmware for automatic configuration? If so, where can I please change it? Thank you.