iNavFlight/inav #4757: plane oscillates wildly in nav modes (rth, 3d cruise)

MrLAtt, 2019-05-27:
My plane flys fine in passthrough and acro modes (after having done a 30 minute long autotune), however if i switch into any of the nav modes the plane starts to oscillate very badly mostly on the pitch axis. (its hard to tell if the erratic movement of the pitch axis is affecting the roll axis or if it is also oscillating on the roll axis)

I tried auto tuning again but it didnt help, i tried lowering the P and I gains also to no avail. attached is an image of my current tuning page

I thought about about changing the barometer settings but figured i may do more bad than good. Also i have never had this probelm before with any of my other inav builds.

[image]

closed 2019-05-30 by MrLAtt

trif55, 2019-06-01:
I have a lot of roll Oscillation on my flying wing, did you work out what to change?

wx4cb, 2019-06-01:
i would guess lower P. or fly it in manual and run an auto tune