iNavFlight/inav #4766: Feature implementation change request: Loiter change

V8Darren, 2019-05-29:
I’d like to recommend a change to the way loiter change is implemented as I think the current method is a bit ambiguous, is a waste of a switch, and could be implemented in a much cleaner way.

I think loiter change could have been better implemented using the yaw input and a couple of settings in configurator. In the bottom right list in the configuration page (where switches like permanently enable airmode are located), I would place an option for “Use yaw to change loiter direction” [on/off]. In the CLI there would be an additional 2 commands:

fw_nav_loiter_enable_change_direction { YES, NO }
# This is the command that the switch in the configuration changes. If yes, the yaw control input can be used to change the loiter direction.

fw_nav_loiter_direction = { LEFT, RIGHT, BEST_DIRECTION }
# This would set the default loiter direction to left or right circles, or use the current state of the aircraft to make the best decision on which way to turn (what I assume iNav does by default currently). This will be used regardless of if the yaw to change direction is enabled.

hali9, 2019-05-30:
Did You try cli command 
`fw_loiter_direction = YAW`
?
https://github.com/iNavFlight/inav/pull/4036#issuecomment-444053858

V8Darren, 2019-06-03:
That works perfectly. But still wonder what's the point of the switch when this works so well.

closed 2019-06-03 by V8Darren

hali9, 2019-06-04:
So that you have the choice, someone prefers to use yaw sticks and someone else prefers to have a switch.

wawanlost, 2022-06-20:
Ketika dipilih yaw,waktu rth fixed wing memutar kanan,kiri atau acak?

hali9, 2022-06-21:
Jika Anda telah mengatur fw_nav_loiter_direction = YAW maka secara default pesawat akan berbelok ke kanan selama RTH / PosHold.
Tetapi jika Anda memindahkan tongkat yaw ke kiri lebih dari setengah saat PosHold aktif, pesawat akan berbelok ke kiri saat RTH dipicu.