iNavFlight/inav #4770: CRSF/iNav/F405-Wing FC: Should I see more than just the 5 telemetry sensors?

padcom, 2019-05-30:
Using the Matek F405 Wing FC with iNAV 2.1.0 firmware, Crossfire 2.93 with MicroRX I see on my Taranis only the following sensors:

- GPS (reported correctly)
- Ptch (reported correctly)
- 1RSS (don't know if the value is correct but it seems to be)
- RxBT (value constantly rises over 100V)
- FM (currently reading !ERR)

When I try to discover other sensors nothing new comes up.

I wonder if there is a setting I missed in iNAV configurator?

I have the "Telemetry" switch to ON, CRSF receiver configured on UART3 (crsf needs a full serial port and since I am using UART2 for CRSF SmartAudio for my VTX I think it is a sensible choice to use Serial 3 for the CRSF RX), I also have GPS on UART5 and additionally I am using MSP on UART4 (with a bluetooth to serial adapter).

Thanks for helping out in advance!

wx4cb, 2019-05-30:
yes you should. make sure you are on the current crossfire version, you should at least see the crossfire sensors (RFMD, TQLY and the like)

teckel12, 2019-05-31:
@padcom As @wx4cb said, you should see many other telemetry sensors.

advancetom, 2019-05-31:
try Beta firmware from Crossfire Version 2.95 and it will work  :)

padcom, 2019-06-01:
So.. I have checked 3 different flight controllers (F405 Wing, F411 Wing - both showing only 5 sensors while the CRSF Micro v2 is being connected to the FC with a proper GPS fix) and Omnibus F4 AIO (very popular I hear) which only shows the 1RSSI sensor.

I have upgraded the CRSF system to the latest beta firmware as was suggested but nothing changed.

Any idea what should I try to do next? What kind of information can I give you guys to help me solve this problem?

padcom, 2019-06-01:
Maybe that'll shed some light on what's going on - here's a dump of my settings. They should be all-default as I just flashed the latest RC iNAV onto my F405-WING FC:

```
# dump

# version
# INAV/MATEKF405SE 2.2.0 May 26 2019 / 19:56:35 (a713b63e4)
# GCC-7.3.1 20180622 (release) [ARM/embedd 0 0 0 0 0 0 0
logic 3 0 0 0 0 0 0 0
logic 4 0 0 0 0 0 0 0
logic 5 0 0 0 0 0 0 0
logic 6 0 0 0 0 0 0 0
logic 7 0_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
eper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
be15200 0 115200
serial 2 64 115200 115200 0 115200
serial 3 1 57600 115200 0 115200
serial 4 2 115200 115200 0 115200
se0::C:0
led 20 0,0::C:0
led 21 0,0::C:0
led 22 0,0::C:0
led 23 0,0::C:0
led 24 0,0::C:0
led 25 0,0::C:0
led 26 # mode_color
mode_color 0 0 1
mode_color 0 1 11
mode_color 0 2 2
mode_color 0 3 13
mode_color 0 4 10
mode_color 0 5 3r 4 3 13
mode_color 4 4 10
mode_color 4 5 3
mode_color 5 0 9
mode_color 5 1 11
mode_color 5 2 2
mode_color 5 3 
aux 8 0 0 900 900
aux 9 0 0 900 900
aux 10 0 0 900 900
aux 11 0 0 900 900
aux 12 0 0 900 900
aux 13 0 0 900 900
aux0 900 0 0
adjrange 10 0 0 900 900 0 0
adjrange 11 0 0 900 900 0 0

# rxrange
rxrange 0 1000 2000
rxrange 1 10000 0
wp 6 0 0 0 0 0 0
wp 7 0 0 0 0 0 0
wp 8 0 0 0 0 0 0
wp 9 0 0 0 0 0 0
wp 10 0 0 0 0 0 0
wp 11 0 0 0 0 0 0
wp 12 0  0 0 0 0
wp 34 0 0 0 0 0 0
[... 181 lines cut ...]
set vtx_low_power_disarch = 7
set fw_ff_pitch = 50
set fw_p_roll = 5
set fw_i_roll = 7
set fw_ff_roll = 50
set fw_p_yaw = 6
set fw_i_yaw = 1n =  1.000
set fw_iterm_limit_stick_position =  0.500
set dterm_notch_hz = 0
set dterm_notch_cutoff = 1
set pidsuet nav_mc_heading_p = 60
set nav_mc_vel_xy_dterm_lpf_hz =  2.000
set nav_fw_pos_z_p = 40
set nav_fw_pos_z_i = 5
set nav1500
set fw_tpa_time_constant = 0
set rc_expo = 70
set rc_yaw_expo = 20
set roll_rate = 20
set pitch_rate = 20
ty = 0
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH
```

padcom, 2019-06-01:
Guys! I found the problem!!! Gosh... I am so stupid...

In the Taranis telemetry screen there is a checkbox "Ignore instances". Then that's checked then the only sensors available are the ones I have mentioned in this issue.

So for posterity: If you only see a limited number of sensors in Taranis make sure you don't have the "Ignore instances" checkbox checked on Telemetry screen in your Taranis OpenTX configuration.

Case closed :)

closed 2019-06-01 by padcom