iNavFlight/inav #4771: Autoland as a flightmode

Ralf-Os, 2019-05-31:
## Current Behavior
Autoland only works with RTH or failsafe RTH. 

## Desired Behavior
Autoland as a flightmode to activate it when needed. 

## Suggested Solution
Make autoland available as a flightmode

## Who does this impact? Who is this for?
All users 

## Additional context
It is sometimes unsafe to activate autoland always or on failsafe. But sometimes it is no Problem to let craft Land by itself. 
2nd use case is landing at desired Position from eg Position hold and not only at RTH point. 
Just loosing some altitude when way too high is another szenario

willbicks, 2019-06-20:
I will second this request. I'm interested in doing autonomous waypoint missions and being able to land at a specific point would be a critical feature. I would love to see this implemented if possible!

sam2b, 2019-07-02:
I too would like this feature.

closed 2019-09-14 by stale[bot]