iNavFlight/inav #4774: Barometer on 2.1.0

ftpn, 2019-06-01:
Hi, i have a SPRACINGF3EVO with a working mbp280 barometer using inav 1.7.1 , to get it working i had to make the following changes to "src/main/target/SPRACINGF3EVO/target .h"
```
#define USE_I2C
//#define I2C_DEVICE (I2CDEV_1)
#define I2C_DEVICE (I2CDEV_2)   // this line changed

#define I2C2_SCL                PA9   // this line added
#define I2C2_SDA                PA10   // this line added
```
i tried to make the same changes to inav 2.1.0 but my barometer no longer works.
I noticed a couple of changes to "src/main/target/SPRACINGF3EVO/target .h", like

```
#define BARO_I2C_BUS            BUS_I2C1 //on line 51 and
#define USE_I2C
#define USE_I2C_DEVICE_1` //on lines 96, 97.
```

The changes i copied from a guy called anbello from rcgroups and it worked so far.
I dont have the knowledge to make it work on the new version.
Any help is appreciated.

teckel12, 2019-06-01:
@Have you tried 2.1.0 without any changes at all?  Maybe barometer support wasn't in 1.7.1 but has since been added?  I've used the SP Racing F3 EVO's barometer without any modification.

ftpn, 2019-06-01:
> @ftpn Have you tried 2.1.0 without any changes at all? Maybe barometer support wasn't in 1.7.1 but has since been added? I've used the SP Racing F3 EVO's barometer without any modification.

teckel12 im using a clone without a on-board barometer or without a I2C port, i added a external barometer to UART1 with the modifications remapping UART1 to I2C, from what i understand from the changes, don`t quote me on that.

teckel12, 2019-06-01:
I was using an on-board barometer. I also no longer fly with that FC so I can't really help.

TimelessNL, 2019-06-30:
I'm working on a F3 board (Micro 32bits F3 EVO V2.0)
[image]
And also trying to get I2C working. I took a brief look at the code and it seems that all sensors use BUS_I2C1 while you're changing the pins for BUS_I2C2.

Could you give this a try?
```
#define USE_I2C
#define USE_I2C_DEVICE_1
#define I2C1_SCL                    PB9       
#define I2C1_SDA                    PB10 
```

TimelessNL, 2019-07-02:
**Solution:**
Ok, the following works (_kinda_):
```
#define USE_I2C
#define USE_I2C_DEVICE_1
#define I2C1_SCL                    PA14  //TX UART2    
#define I2C1_SDA                    PB15 //RX UART2
```
**Extra info:**
UART2 is chosen since had I already soldered the SDA/SCL sensor wires to these pins.
I did however test UART1 as well but was unable to get i2c to work in this port.
There is however something strange, at first I though there was something with my sensors and/or pull_up resistors, since I got massive (+10k) i2c errors and only my _BMP180_ was detected while my _QCM5883L_ did not. Also the CPU load is over 600%.

**Problem:**
After some _try & error_ I discovered that this had something to do with the RX setting :/.
The SPRACINGF3EVO's default RX mode is set to PPM:
`#define DEFAULT_RX_TYPE RX_TYPE_PPM`
When I set it to anything else (in my case iBUS) i2c works perfectly _(zero i2c errors)_ and all attached sensors are detected.

**Question:**
Does anyone else experience "_RX PPM mode_" issue with i2c?
I do not have any PPM receiver connected though,
to be more precise there isn't any receiver attached at all (yet).
But find it strange that if there is no receiver present that iNav is on the verge of crashing.

closed 2019-09-14 by stale[bot]