iNavFlight/inav #4776: SURFACE mode doesn't work

georgekucher, 2019-06-01:
## Current Behavior

Quadcopter doesn't hold fixed altitude above the surface in SURFACE mode. Log is attached. Please help.
[LOG00067.TXT](https://github.com/iNavFlight/inav/files/3243690/LOG00067.TXT)
## Steps to Reproduce

1.Arm and take off
2.Enable SURFACE mode

## Expected behavior

Copter is expected to hold altitude above the ground when SURFACE mode is being enabled.

## Suggested solution(s)

Maybe it is disabled for my target..

## Additional context

[Dump](https://pastebin.com/DcHQQGAg)

---

- FC Board name and vendor: INAV/MATEKF722 2.2.0 May 26 2019 / 19:50:46 (a713b63e4)
- INAV version string: GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

georgekucher, 2019-06-01:
I have mentioned it before
https://github.com/iNavFlight/inav-rangefinder/issues/6#issuecomment-487483702

digitalentity, 2019-06-01:
Surface more is a modifier for altitude hold mode. It has to be enabled together with ALTHOLD or at earlier time.

georgekucher, 2019-06-01:
I've tried this way as well. See my first comment.
Still not working

georgekucher, 2019-06-01:
Or do you mean that firstly I should enable SURFACE and only after this ALTHOLD?

digitalentity, 2019-06-01:
`SURFACE` should be enabled **before or at the same moment** as `ALTHOLD`. In your log your copter is not in surface mode, but in regular altitude hold.

digitalentity, 2019-06-01:
`SURFACE` is not a mode. It changes the behavior of `ALTHOLD`

digitalentity, 2019-06-01:
Best - take off in ALTHOLD + SURFACE mode

georgekucher, 2019-06-01:
Will try and reply in 1 hour. BTW, why max sonar range is only 60 cm?

digitalentity, 2019-06-01:
Crappy sonar hardware. INAV's limits for rangefinder/surface tracking
`max_surface_altitude` (2m default) - max rangefinder reading INAV will accept as a valid.
`nav_max_terrain_follow_alt` (1m default) - max altitude INAV will allow in terrain following mode.

georgekucher, 2019-06-01:
Hm.. I'm using us100 + arduino. May it be limits in sketch?

digitalentity, 2019-06-01:
On a 3-5" quad US-100 gives 60-80cm range unless you use a very good soft-mounting (soft rubber) to isolate US100 from vibrations.

georgekucher, 2019-06-01:
I was testing range without arming. It just displays -1 write after exceeding 60 cm. That's why I think it is limited somewhere in code.
Can surface mode be always enabled? Or it can brake some other behavior like RTH or POSHOLD or GSC or any other?

digitalentity, 2019-06-01:
That could be limited in Arduino sketch. INAV doesn't have this 60cm limit.
SURFACE mode shouldn't interfere with RTH, but it **will** affect POSHOLD and GSC

georgekucher, 2019-06-01:
[Here](https://github.com/iNavFlight/inav-rangefinder/blob/a947f01d4ba065536c7273b7ddcdf526b5d453aa/inav_rangefinder.ino) is the sketch I use.

For POSHOLD and GSC it (SURFACE) should be enabled simultaneously as well?

georgekucher, 2019-06-01:
@digitalentity It is definitely limited in the sketch. Playing with these vars PULSE_TO_CM 500,  MAX_RANGE 500. Have no idea what is what, behavior is strange, but still)))

georgekucher, 2019-06-02:
OK. I finally sorted our this sketch.
So. PULSE_TO_CM is not used for calculations of distance, at least for US 100. It is used (multiplied by MAX_RANGE) only for cutting off max range.
Another thing is that cutting off has a strange behavior. When MAX_RANGE is 300 (which is 3 meters not 4 how is written in cooment) it works only up to 60 cm. Vert weird for me. But if I put MAX_RANGE

georgekucher, 2019-06-02:
https://github.com/iNavFlight/inav-rangefinder/issues/6#issuecomment-498005047

@digitalentity Which behavior should I expect in POSHOLD and GSC modes (with SURFACE modifier)?
P.S. Sorry for so many questions. This feature is poor documented yes.

daniel214, 2019-06-03:
Summarizing discussions from chat, I think the dependence on ALTHOLD is a major usability trap that could be resolved by quietly making SURFACE enable the altitude-hold code.

ie pseudocode
`is_altitude_hold_enabled = isBoxMode(SURFACE) || isBoxMode(ALTHOLD)`

It is not clear to me what the benefit of keeping SURFACE as a modifier dependent on ALTHOLD, especially if you can't toggle it in flight.

georgekucher, 2019-06-03:
I agree that this is not very obvious way to switch SURFACE.
BTW here is an update 
https://github.com/iNavFlight/inav-rangefinder/issues/6#issuecomment-498414007

Seeelefant, 2019-06-30:
I kind off have the same problem, sensor is a TFMini, however I see no effect of switching on surface mode even together with AltHold.  Inav is version 2.2.
Please advice.
[LOG00109.zip](https://github.com/iNavFlight/inav/files/3343343/LOG00109.zip)

closed 2019-11-18 by stale[bot]

pDR8iLQ4go, 2020-04-10:
I‘m testing the feature SURFACE as well. But the behavior differs. I had activated SURFACE together with POSHOLD. Seems to work as expected. But sometimes I‘d like to activate POSHOLD without SURFACE. How to parameterize that in INAV configurator?

georgekucher, 2020-04-10:
Is anyone going to look into this issue?

pDR8iLQ4go, 2020-04-10:
@georgekucher: Do you use this feature? I‘m using the Matek sensor, see https://www.rcgroups.com/forums/showthread.php?3572803-Matek-Lidar-and-INAV. It works more or less. But I do not know how to parameterize SURFACE in combination with ALTHOLD or POSHOLD.

georgekucher, 2020-04-12:
Yes I have the same sensor.

> `SURFACE` should be enabled **before or at the same moment** as `ALTHOLD`. In your log your copter is not in surface mode, but in regular altitude hold.

I'm enabling it (SURFACE) **before** enabling ALT/POS HOLD and it works.

Enabling SURFACE+ALT/POS HOLD simultaneously **didn't** work for me.
So that it requires more TX switchers to be used for this manipulations((

pDR8iLQ4go, 2020-04-12:
In my case, it's exactly the opposite.

This setup works:
[image]
"Works" means: In case of activating POSHOLD&SURFACE, the copter sinks and keeps hovering at the level of approx. 100cm.

This setup doesn't works:
[image]
"Doesn't work" means: My copter will sink slowly to the ground without hovering at. When the copter reaches the ground, he rises slowly. This will repeat: sink - rise - sink - rise - ...

georgekucher, 2020-04-12:
Hmm.. Maybe devs changed something.
I have following config
[image]

pDR8iLQ4go, 2020-04-12:
I assume it will take a while for me to understand what you are doing. How many positions has the switch which is linked to channel 6? It seams yo control nearly everything with this switch. Right?

georgekucher, 2020-04-12:
I've mixed 2 3-pos. So 9 positions.
Yes I control all flight modes with this channel

daniel214, 2020-04-12:
It was my understanding that SURFACE mode has to be enabled before (or perhaps simultaneously with) ALTHOLD for the surface following mode to activate. SURFACE has no function when activated separately-- it's purely a modifier for ALTHOLD.
See my comment far above about the usability problem this causes because it is difficult to understand.

georgekucher, 2020-04-12:
> It was my understanding that SURFACE mode has to be enabled before (or perhaps simultaneously with) ALTHOLD for the surface following mode to activate. SURFACE has no function when activated [...]

Yes, I totally agree with you. Further more, the same should be done for POSHOLD+SURFACE

daniel214, 2020-04-12:
Wait... does this work? You can get into SURFACE mode while in POSHOLD mode without ALTHOLD on?

pDR8iLQ4go, 2020-04-12:
That's my aim. But it does not work. POSHOLD and ALTHOLD works, but without the lidar surface feature. And I thought I can activate the lidar surface feature separately.

I assume that I have to try mixing channels to achieve that. I up to now I haven't used the channel mix feature.

georgekucher, 2020-04-12:
> Wait... does this work? You can get into surface following mode while in POSHOLD mode without ALTHOLD on?

Yes

georgekucher, 2020-04-12:
Actually I can get into POSHOLD while it SURFACE

daniel214, 2020-04-12:
Try it this way. Using two three-pos switches,
Switch 1:
Pos A) SURFACE+ALTHOLD
Pos B) Nothing
Pos C) ALTHOLD

Switch 2:
Nav modes, however you like.

georgekucher, 2020-04-12:
> Try it this way. Using two three-pos switches,
> Switch 1:
> Pos A) SURFACE+ALTHOLD
> Pos B) Nothing
> Pos C) ALTHOLD

This is my config now

pDR8iLQ4go, 2020-04-12:
I have the budget FS-i6. With the 10 channel update and some modified switches.
[image]

I try to use the channel mix feature to solve the task.

georgekucher, 2020-04-12:
I have the same. Only with 16 channels and a bit different switches upgrade

pDR8iLQ4go, 2020-04-12:
Great! May you be so kind and tell me how to parameterize the mix? I‘d like to use two of my 3 position switches to do what you did. Setup has to be performed in the radio. Right?

daniel214, 2020-04-12:
I'm a bit confused why this overly complicated setup is necessary. For example, if you drop the need to fly in althold modes without poshold active (when nav_user_control_mode is set to ATTI, which is the default, you get angle-mode control while in POSHOLD anyway), you could do this:

Switch 1: Position hold modes
Position A) POSHOLD+ALTHOLD
Position B) Nothing
Position C) POSHOLD+ALTHOLD+SURFACE

Switch 2: Autonomous navigation modes
Position A) NAV WP
Position B) Nothing
Position C) NAV RTH

I don't have the ability to test this right now, but as far as I can tell, NAV WP and NAV RTH override ALTHOLD anyway to follow their programmed waypoints or return-to-home behavior.

georgekucher, 2020-04-13:
> I'm a bit confused why this overly complicated setup is necessary. For example, if you drop the need to fly in althold modes without poshold active (when nav_user_control_mode is set to ATTI, which [...]

The thing is that I don't want to drop flying in ALTHOLD. This is why I mixed switches into 9pos

georgekucher, 2020-04-13:
> Great! May you be so kind and tell me how to parameterize the mix? I‘d like to use two of my 3 position switches to do what you did. Setup has to be performed in the radio. Right?

Yes. In radio. You should go to TX setup and assign 2 switches to a channel. In my case this is Channel6=SWB+SWC. The firmware of the TX should have this options.
[image]
[image]

georgekucher, 2020-04-13:
[image]

pDR8iLQ4go, 2020-04-13:
Have you ever seen this strange behavior with SURFACE & POSHOLD and full throttle?
https://youtu.be/0SNdMbonubk

georgekucher, 2020-04-13:
Hm..
No never.

pDR8iLQ4go, 2020-04-13:
I'm despairing. Have you performed the "Optic Flow Calibration"? I did that several times on different surface (inside, outside, lawn, wood, ...). The result (value of "Scale") is always different. But the calibration seems to have impact on the result of the debug signals. Especially on Debug 1. Currently, it looks like that:
[image]

Should be ok, but the Sonar sensor signal looks not that good:
[image]

georgekucher, 2020-04-13:
> Have you performed the "Optic Flow Calibration"?

No I didn't.

> Should be ok, but the Sonar sensor signal looks not that good:

My performs very likely.

daniel214, 2020-04-13:
This has now veered far off topic, but IR rangefinders can have difficulty in full sunlight.

pDR8iLQ4go, 2020-04-13:
Thanks anyway...

ElFab, 2021-07-15:
I know this is a necro but i just want to say that the current behaviour is really bad, even more it's dangerous....  Currently when i switch from poshold to surface + PosHold it just go acro ( i understand that now reading you all ). Now i'm trying to make this work fine but i don't realy found a nice solution.
 I'm with "Use mid. throttle for ALTHOLD" setting enabled because i want to switch to Poshold without the need to hover before (without that i would not know where the neutral Poshold throttle would be). The need to switch to acro (or maybe angle would work but same problem) for activate surface is dangerous. My quad is hovering in angle or acro at around 1250 (inav throttle value), 1500 (even for a short time) make it jump high,  i don't know what Surface will do that high. Even if surface work if activated from a high height each time i will want to use surface i will need to be really carefull.
I think for now i will just forget this feature, it need to be a mode in itself and activate PosHold (or at least have the ability to be activated after the mod it modify).

In fact what it should be and i want to know if you agree is :
- A modifier for PosHold(or altHold) (can be activate after would less mater in this case because you'll always use it in nav modes).
- Automatic activation if under 200cm 
- Automatique desactivation if at max height and throttle max (back to PosHold) with z velocity P reduced for a while (smoothely go back to normal or direct if setPoint < 0) 
- Land if at min height and throttle at 0
I'll take a look at source code maybe it's not possible i'm new in the drone's world.
I would appreciate to know what you think about that.