iNavFlight/inav #4777: auto slowdown speed when near the desire point

sincoder, 2019-06-01:
## Current Behavior
quadcopter fly to the waypoint (or home point) at NAV speed

## Desired Behavior
quadcopter auto slowdown it's speed and stop at the desire point
## Suggested Solution
quadcopter fly near the target point ,for example, 14 meters and NAV speed is 14m/s .FC set desire speed to 13m/s ,and when distance become 13 m/s, FC set desire speed to 12 m/s and so on, the quad will fly more gently and stop almost the desire point. 

## Who does this impact? Who is this for?
quadcopters

## Additional context
when I RTH at high speed , 15M/s + , the quad would be too busy to HOLD it's position above home point. It fly forward and backward, sometimes circle around. and takes about 10 seconds to find the right position and calm down .

sincoder, 2019-06-01:
Don't suggest me RTH at 1m/s ...

digitalentity, 2019-06-01:
This is exactly how it works. Distance to target is translated to target velocity by `PosXY P`. Reduce it to have a more gentle approach.

closed 2019-06-01 by digitalentity