iNavFlight/inav #4779: SBUS and FPORT not working with R9Slim+ (on 3 different FCs)

Fredisk, 2019-06-01:
## Current Behavior
FPORT and SBUS don't work right with KakuteF7, FF F35, Kakute F4V2. On every FC I tried different versions (also pre. 2.0.0 if possible)

here is a image of what I mean:
https://snag.gy/5cw72s.jpg
Sometimes it is that synchronized and sometimes it feels total random.

## Steps to Reproduce
Just a usual R9Slim+ with SBUS or FPORT and any FC I mentioned above. 

## Expected behavior
SBUS or FPORT working

## Suggested solution(s)
After around 2hrs of research I found some solutions but they aren't available for my FC.... 

## Additional context
I also tried playing around with the "set sbus_sync_interval" command but no success.

- FC Board name and vendor: KakuteF7, FF F35 Lightning, Kakute F4V2
- INAV version string: On FF F35 I tested every Version, on KakuteF7 there was only one version (2.something) and F4V2 I also tested 1.9.1 and 2.+
[image]

Fossdal, 2019-06-19:
I have an Kakute F7-SE and an Kakute F4v2 and use R9Slim+ connected with SBUS. it works fine.
to diagnose you need to post an dump of your settings and tell witch pins you have connected SBUS to

DzikuVx, 2019-06-19:
I have 3 Slim+ connected to 3 different FCs and all work like they should be. I expect this is user configuratiopn error

closed 2019-06-19 by DzikuVx