iNavFlight/inav #4783: 2.1 RC1 - RGB LED Strip Control broken on Matek F405SE

b14ckyy, 2019-06-02:
## Current Behavior
RGB LED Control seems not to work on Matek F405SE (Wing) - LEDs just randomly flashing or flickering.

## Steps to Reproduce
1. Setup RGB LEDs on a F405SE Wing with INAV 2.2 RC1
2. Power on board
3. if some LEDs will power up at all (depends on the LEDs) they will randomly shining or flickering

4. all fine when flashing back to 2.1 
5. same with scratch setup LEDs or restoring LED settings from 2.1

## Expected behavior
LED control behavior should be the same as in 2.1?

## Suggested solution(s)
Check for LED control output on Matek F405SE Wing

## Additional context
Tested and confirmed this with 2 flight controllers and 2 known good sets of RGB LEDs. 

INAV 2.1 on spare board and spare LED stripe: https://youtu.be/7DwqFvHWK78
INAV 2.2 on same board and same LED stripe: https://youtu.be/JyKD9zjN_UE

---

- FC Board name and vendor: INAV/MATEKF405SE 2.2.0 May 26 2019 / 19:56:35 (a713b63e4)
- INAV version string: GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

b14ckyy, 2019-06-05:
Still the same in RC2

digitalentity, 2019-06-08:
Fixed by #4805. Closing

PR #4805 (2019-06-09): Reduce LEDSTRIP timer clock to match CPU clock
LEDSTRIP timer frequency has value to of 24MHz. On F3 (timer clock 72MHz) this yields prescaler of 3 and everything works normally.

However on F405 (timer clock either 168 or 84 MHz) this may yield a non-integer prescaler of 3.5 on certain timers. 
INAV 2.1 used truncation and yielded LEDSTRIP carrier frequency of 933KHz.
INAV 2.2 uses proper rounding and generates LEDSTRIP carrier of 700KHz.

WS2811 expects carrier of 800KHz and based on individual chip tolerances may work (or not work) with different frequency.

This PR lowers timer frequency t [... cut]
closed 2019-06-08 by digitalentity