iNavFlight/inav #4793: Change boardalignment on switch and different mixing profiles

printme3daachen, 2019-06-04:
Hi, I've build a flying wing with VTOL capabilities and it's working great so far. I can fly it perfectly as a bicopter so far. 
For translation to flying like a plane I have two problems which need to be solved.
In the adjustments tab I found a setting which would change my board alignment angle. But I need it to change it to a fixed value. Right now I can only change it by using a switch and and holding it until it fits. Tht's very complicated to do while flying and very unprecise. Is there a way to set the angle to two different values depending on a switch state?

Second issue is that I need to change the mixing while flying with the same switch. When switching from vertical to horizontal flight, the yaw and roll axis change and therefore I would need to change the mixing. I thought this could be done by saving two mixes in two profiles. Unfortunately this does only work with things like the PIDs right now. Would it be possible to add this feature also for multiple mixes?

Here are some pictures of the current state. Very experimental but looks promissing to me.

[image]
[image]

hali9, 2019-06-05:
When it comes to servo, you can use condition logic in smix. When it comes to engines, this is not yet done.
> In the adjustments tab I found a setting which would change my board alignment angle. But I need it to change it to a fixed value

I have not found such a function anywhere. And it would be very desirable for VSTOL.
Moving 'board angle' to profile, this is a solution but probably not correct.

I think a VSTOL mode should be created, which would change the settings to other new ones defined in the variable cli (vstol_angle = 90 ?) in some maybe predefined time in cli (vstol_time = 2 seconds ?). It could also change the profile. 
For example when ''board angle' = 0
From 0-30 use profile 1, 30-60 use profile 2, 60-90 use profile 3.

But it is not so easy because the flight controller can be mounted differently and it is not known which angle to change it would also need to be defined (vstol_direction ?), besides would have to change the type from  fixedwing to multirotor. Unless You use fixedwing also for hovering. Maybe you can share your `cli diff all`.

Changing the profile is probably possible using adjustment also.

printme3daachen, 2019-06-05:
Thanks for your reply. I don't know anything about smix, but if it doesn't work with motors it propably wont work for me, because I would also need to change their mixing to from yaw to roll. 

> I have not found such a function anywhere. And it would be very desirable for VSTOL.
> Moving 'board angle' to profile, this is a solution but probably not correct.

It can be found in the adjustments tab. 
[image]

A VSTOL mode would be great, but for me I'd rather prefer a quick and easy solution by setting up to profiles including two seperate mixes of motors and servos plus two different board alignment settings and of course to different PID tunings (the last one is already possible).
The idea of switching between the profiles by change the settings slowly until the desired value is reached sounds great, but also very complicated to program and configure because you would also need to slowly change the PIDs while transitioning. Is that what you wanted to solve with the different profiles (0-30,30-60,60-90)? That would be a pretty easy way, but is it worth the work to have 3 profiles including one 'inbetween' profile instead of having two?

I would rather have the two profile soultion integrated pretty soon instead of waiting very long and have something with build in transition. Also, for the two profiles the only things that would have to be implemented are saving not only PIDs but also a mix in the profile and save the board alignment settings in the profile. I don't know how easy this would be, but I think much simpler than having an interpolation running for a time x. This would also need to have the two profiles, so implementing those settings into the profiles is definitly the first step.

Here is my diff all ;)

`# diff all

# version
# INAV/OMNIBUSF4PRO 2.1.0 Feb 25 2019 / 16:53:27 (65b0ec142)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

# reset configuration to default settings
defaults noreboot

# resources

# mixer
mmix 0  1.000  1.000  0.000  0.000
mmix 1  1.000 -1.000  0.000  0.000

# servo mix
smix 0 3 2 50 0
smix 1 3 1 50 0
smix 2 4 2 -50 0
smix 3 4 1 50 0

# servo
servo 3 500 2500 1530 -100
servo 4 500 2500 1550 100

# feature
feature -TX_PROF_SEL
feature SOFTSERIAL
feature TELEMETRY
feature PWM_OUTPUT_ENABLE

# beeper

# map

# serial
serial 20 1 115200 115200 0 115200
serial 0 64 115200 115200 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100
aux 1 1 2 900 1300
aux 2 28 1 1700 2100

# adjrange

# rxrange

# temp_sensor

# osd_layout

# master
set acc_hardware = MPU6000
set acczero_x = 65
set acczero_y = 3
set acczero_z = -42
set accgain_x = 4052
set accgain_y = 4058
set accgain_z = 4036
set mag_hardware = NONE
set baro_hardware = BMP280
set pitot_hardware = NONE
set receiver_type = SERIAL
set serialrx_provider = SBUS
set motor_pwm_rate = 8000
set motor_pwm_protocol = DSHOT300
set failsafe_procedure = DROP
set align_board_roll = -15
set align_board_pitch = 820
set platform_type = AIRPLANE
set model_preview_type = 8

# profile
profile 1

set fw_p_pitch = 70
set fw_i_pitch = 15
set fw_ff_pitch = 10
set fw_p_roll = 35
set fw_i_roll = 25
set fw_ff_roll = 10
set fw_p_yaw = 33
set fw_i_yaw = 30
set fw_ff_yaw = 10
set rc_expo = 0
set rc_yaw_expo = 0
set manual_rc_expo = 0
set manual_rc_yaw_expo = 0

# profile
profile 2

# profile
profile 3

# battery_profile
battery_profile 1

# battery_profile
battery_profile 2

# battery_profile
battery_profile 3

# restore original profile selection
profile 1
battery_profile 1

# save configuration
save`

closed 2019-08-18 by stale[bot]

gavinlowk, 2019-11-09:
> 
> 
> Thanks for your reply. I don't know about condition logic in smix, but if it doesn't work with motors it propably wont work for me, because I would also need to change their mix from yaw to roll.
> 
> > I have not found such a function anywhere. And it would be very desirable for VSTOL.
> > Moving 'board angle' to profile, this is a solution but probably not correct.
> 
> It can be found in the adjustments tab.
> [image]
> 
> A VSTOL mode would be great, but for me I'd rather prefer a quick and easy solution by setting up two profiles including two seperate mixes of motors and servos plus two different board alignment settings and of course two different PID tunings (the last one is already possible).
> The idea of switching between the profiles by changing the settings slowly until the desired value is reached sounds great, but also very complicated to program and configure because you would also need to slowly change the PIDs while transitioning. Is that what you wanted to solve with the different profiles (0-30,30-60,60-90)? That would be a pretty easy way, but is it worth the work to have 3 profiles including one 'inbetween' profile instead of having two?
> 
> I would rather have the two profile soultion integrated pretty soon instead of waiting very long and have something with build in transition. Also, for the two profiles the only things that would have to be implemented are saving not only PIDs but also a mix and the board alignment settings in the profile. I don't know how easy this would be, but I think much simpler than having an interpolation running for a time x. This would also need to have the two profiles, so implementing those settings into the profiles is definitly the first step.
> 
> Here is my diff all ;)
> 
> `# diff all
> # version
> # INAV/OMNIBUSF4PRO 2.1.0 Feb 25 2019 / 16:53:27 ([65b0ec1](https://github.com/iNavFlight/inav/commit/65b0ec142c5b27bb5f0e8ea14f8d0627cff7ca76))
> # GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
> # reset configuration to default settings
> 
> defaults noreboot
> # resources
> # mixer
> 
> mmix 0 1.000 1.000 0.000 0.000
> mmix 1 1.000 -1.000 0.000 0.000
> # servo mix
> 
> smix 0 3 2 50 0
> smix 1 3 1 50 0
> smix 2 4 2 -50 0
> smix 3 4 1 50 0
> # servo
> 
> servo 3 500 2500 1530 -100
> servo 4 500 2500 1550 100
> # feature
> 
> feature -TX_PROF_SEL
> feature SOFTSERIAL
> feature TELEMETRY
> feature PWM_OUTPUT_ENABLE
> # beeper
> # map
> # serial
> 
> serial 20 1 115200 115200 0 115200
> serial 0 64 115200 115200 0 115200
> # led
> # color
> # mode_color
> # aux
> 
> aux 0 0 0 1700 2100
> aux 1 1 2 900 1300
> aux 2 28 1 1700 2100
> # adjrange
> # rxrange
> # temp_sensor
> # osd_layout
> # master
> 
> set acc_hardware = MPU6000
> set acczero_x = 65
> set acczero_y = 3
> set acczero_z = -42
> set accgain_x = 4052
> set accgain_y = 4058
> set accgain_z = 4036
> set mag_hardware = NONE
> set baro_hardware = BMP280
> set pitot_hardware = NONE
> set receiver_type = SERIAL
> set serialrx_provider = SBUS
> set motor_pwm_rate = 8000
> set motor_pwm_protocol = DSHOT300
> set failsafe_procedure = DROP
> set align_board_roll = -15
> set align_board_pitch = 820
> set platform_type = AIRPLANE
> set model_preview_type = 8
> # profile
> 
> profile 1
> 
> set fw_p_pitch = 70
> set fw_i_pitch = 15
> set fw_ff_pitch = 10
> set fw_p_roll = 35
> set fw_i_roll = 25
> set fw_ff_roll = 10
> set fw_p_yaw = 33
> set fw_i_yaw = 30
> set fw_ff_yaw = 10
> set rc_expo = 0
> set rc_yaw_expo = 0
> set manual_rc_expo = 0
> set manual_rc_yaw_expo = 0
> # profile
> 
> profile 2
> # profile
> 
> profile 3
> # battery_profile
> 
> battery_profile 1
> # battery_profile
> 
> battery_profile 2
> # battery_profile
> 
> battery_profile 3
> # restore original profile selection
> 
> profile 1
> battery_profile 1
> # save configuration
> 
> save`

Hi printme3daachen
I really love your way to the VTOL in simple profile setting.
Any update you can set VTOL with 2 profile transition ?