iNavFlight/inav #4800: Matek F722-SE no camera image

KiloTango213, 2019-06-06:
Hello there,

I am using a Matek F722-SE FC with iNav and for now everything worked great.
But i have a problem with the VTX und C1 conections.

I connected my vtx transmitter to VBat, Gnd and VTX and my runcam micro eagle to c1 5v and gnd, just like the describtion in the matek manual. 

Problem is I dont get an image to my receiver. 
However if I connect just the video channel directly from camera to vtx transmitter  I get an image.

I played with User 1 and User2 to activate the transmitter and camera wich worked fine, but still no image.

So my question now is: Is there some sort of switch to enable c1 or anything else I missed and could do?

DzikuVx, 2019-06-10:
Check if connecting RX and having radio turned on helps.

KiloTango213, 2019-06-10:
Thanks for the tip. I checked it twice but it didnt fix it.
I talked to Matek Support and within 1 hour and sending some pictures, they told me to check back with the dealer to get a new one. They replacement is on its way to me.

closed 2019-06-10 by DzikuVx

sam2b, 2019-07-02:
@kilotango213 I had the exact same problem and was deemed a defective flight controller as per Matek support. Reach out to them. And make a clear detailed video demonstrating the problem.