iNavFlight/inav #4804: Error opening log: No logs in this file could be parsed successfully

georgekucher, 2019-06-08:
## Current Behavior

## Steps to Reproduce

1. Flash 2.2.0 RC1 or RC2
2. Arm
3. Disarm
4.Try to open log.

## Expected behavior

Log opens

## Additional context

[LOG00001.TXT](https://github.com/iNavFlight/inav/files/3268420/LOG00001.TXT)

https://pastebin.com/5sXHegUk
---

# version
# INAV/MATEKF722 2.2.0 Jun  3 2019 / 14:05:06 (f59251378)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

junwoo091400, 2019-06-10:
Hmmmmmmm any suspicious parts of the code which might have caused this behavior? I don't see it for now..😞

shellixyz, 2019-06-10:
No clue yet I already took a look and tried to fix it but failed

stronnag, 2019-06-10:
2.2 BB Logs work fine here on a flying wing (servos are logged). Or am I misunderstanding the title?

stronnag, 2019-06-10:
The reporter's log above is showing corruption even in the headers. e.g
**Reported Log**
```
H Firmware type:Cleanflight
H FirmwH thr_mid:50
H thr_expo:0
H tpa_rate:0
```
**My Log of 2019-06-19**
```
H Firmware type:Cleanflight
H Firmware revision:INAV 2.2.0 (9ddbe4743) WINGFC
H Firmware date:Jun  9 2019 12:37:32
H Log start datetime:2019-06-10T08:47:48.075+01:00
H Craft name:Wing it!
H P interval:1/32
H minthrottle:1060
H maxthrottle:2000
H gyro_scale:0x3f800000
H motorOutput:1060,2000
H acc_1G:2048
H vbat_scale:110
H vbatcellvoltage:33,35,42
H vbatref:1248
H currentMeter:0,250
H looptime:1000
H rc_rate:100
H rc_expo:70
H rc_yaw_expo:20
H thr_mid:50
```
(this is really 818a04f8d, with the `src/main/sensors/battery.c` "mis-compile" workaround reverted for GCC 9).

As hunks of the headers are missing, this smells more like a platform issue to me.

shellixyz, 2019-06-10:
You understood the title right it seems like there is an issue when the servos are enabled only. I did not experience the issue myself because I don't have a wing available to test this for now but another user have reported the issue on Telegram and his log didn't seem to be corrupted. I don't know if I can share the log itself but I can share the header. He is using a custom build for the OMNINXT F7.

```
H Product:Blackbox flight data recorder by Nicholas Sherlock
H Data version:2
H I interval:32
H Field I name:loopIteration,time,axisRate[0],axisRate[1],axisRate[2],axisP[0],axisP[1],axisP[2],axisI[0],axisI[1],axisI[2],axisD[0],axisD[1],mcPosAxisP[0],mcPosAxisP[1],mcPosAxisP[2],mcVelAxisP[0],mcVelAxisP[1],mcVelAxisP[2],mcVelAxisI[0],mcVelAxisI[1],mcVelAxisI[2],mcVelAxisD[0],mcVelAxisD[1],mcVelAxisD[2],mcVelAxisFF[0],mcVelAxisFF[1],mcVelAxisFF[2],mcVelAxisOut[0],mcVelAxisOut[1],mcVelAxisOut[2],mcSurfaceP,mcSurfaceI,mcSurfaceD,mcSurfaceOut,rcData[0],rcData[1],rcData[2],rcData[3],rcCommand[0],rcCommand[1],rcCommand[2],rcCommand[3],vbat,amperage,magADC[0],magADC[1],magADC[2],BaroAlt,gyroADC[0],gyroADC[1],gyroADC[2],accSmooth[0],accSmooth[1],accSmooth[2],attitude[0],attitude[1],attitude[2],debug[0],debug[1],debug[2],debug[3],debug[4],debug[5],debug[6],debug[7],motor[0],motor[1],motor[2],motor[3],servo[0],servo[1],servo[2],servo[3],servo[4],servo[5],servo[6],servo[7],navState,navFlags,navEPH,navEPV,navPos[0],navPos[1],navPos[2],navVel[0],navVel[1],navVel[2],navAcc[0],navAcc[1],navAcc[2],navTgtVel[0],navTgtVel[1],navTgtVel[2],navTgtPos[0],navTgtPos[1],navTgtPos[2],navSurf[0]
H Field I signed:0,0,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,0,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,0,0,0,0,0,0,0,0,0,0,0,1,0,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1
H Field I predictor:0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,4,9,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,4,5,5,5,8,8,8,8,8,8,8,8,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
H Field I encoding:1,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,1,3,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
H Field P predictor:6,2,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,3,3,3,3,3,3,3,3,3,1,1,1,1
H Field P encoding:9,0,0,0,0,0,0,0,7,7,7,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,8,8,8,8,8,8,8,8,6,6,6,6,6,6,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
H Field H name:GPS_home[0],GPS_home[1]
H Field H signed:1,1
H Field H predictor:0,0
H Field H encoding:0,0
H Field G name:time,GPS_fixType,GPS_numSat,GPS_coord[0],GPS_coord[1],GPS_altitude,GPS_speed,GPS_ground_course,GPS_hdop,GPS_eph,GPS_epv,GPS_velned[0],GPS_velned[1],GPS_velned[2]
H Field G signed:0,0,0,1,1,1,0,0,0,0,0,1,1,1
H Field G predictor:10,0,0,7,7,0,0,0,0,0,0,0,0,0
H Field G encoding:1,1,1,0,0,0,1,1,1,1,1,0,0,0
H Field S name:flightModeFlags,stateFlags,failsafePhase,rxSignalReceived,rxFlightChannelsValid,hwHealthStatus,powerSupplyImpedance,sagCompensatedVBat,wind[0],wind[1],wind[2],IMUTemperature,baroTemperature,sens0Temp,sens1Temp,sens2Temp,sens3Temp,sens4Temp,sens5Temp,sens6Temp,sens7Temp
H Field S signed:0,0,0,0,0,0,0,0,1,1,1,1,1,1,1,1,1,1,1,1,1
H Field S predictor:0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
H Field S encoding:1,1,7,7,7,1,1,1,0,0,0,0,0,0,0,0,0,0,0,0,0
H Firmware type:Cleanflight
H Firmware revision:INAV 2.2.0 () AEGIMBAL
H Firmware date:Jun  9 2019 19:22:39
H Log start datetime:2019-06-09T19:33:57.681+00:00
H Craft name:FlyinGimbal
H P interval:1/5
H minthrottle:1150
H maxthrottle:1850
H gyro_scale:0x3f800000
H motorOutput:1150,1850
H acc_1G:2048
H vbat_scale:111
H vbatcellvoltage:33,35,42
H vbatref:1491
H currentMeter:390,1000
H looptime:250
H rc_rate:100
H rc_expo:70
H rc_yaw_expo:70
H thr_mid:50
H thr_expo:0
H tpa_rate:0
H tpa_breakpoint:1500
H rates:60,60,50
H rollPID:40,35,24
H pitchPID:45,35,24
H yawPID:85,45,0
H altPID:50,0,0
H posPID:65,0,0
H posrPID:40,15,100
H levelPID:20,15,75
H magPID:60
H velPID:100,50,10
H yaw_p_limit:300
H yaw_lpf_hz:0
H dterm_lpf_hz:75
H dterm_notch_hz:0
H dterm_notch_cutoff:1
H deadband:5
H yaw_deadband:5
H gyro_lpf:0
H gyro_lpf_hz:90
H gyro_notch_hz:0,0
H gyro_notch_cutoff:1,1
H acc_lpf_hz:15
H acc_hardware:7
H baro_hardware:6
H mag_hardware:2
H serialrx_provider:10
H motor_pwm_protocol:7
H motor_pwm_rate:16000
H debug_mode:1
H features:810093762
H waypoints:0,0
H acc_notch_hz:0
H acc_notch_cutoff:1
H gyro_stage2_lowpass_hz:175
H dterm_setpoint_weight: 1.000
H pidSumLimit:500
H axisAccelerationLimitYaw:10000
H axisAccelerationLimitRollPitch:0
```

stronnag, 2019-06-10:
So why are my flying wings 2.2 logs OK ?

shellixyz, 2019-06-10:
Well that proves that the issue is not only due to the servos enabled. The best way to find what the issue is is probably to find what makes the BB viewer reject the logs

stronnag, 2019-06-10:
BB Logs work with servos enabled (flying wings) on F3 and F4 here, the common failing platform is F7 (which I can't test). Servos don't seem to be an issue for either F3 or F4.

digitalentity, 2019-06-10:
It could be the issue with SD-Card (as usual). With servos BB header is a bit longer and if data loss occurs there things might get corrupt beyond BB Log Viewer ability to recover. Try a different SD-Card.

stronnag, 2019-06-10:
It's also the case that `blackbox-decode` fails to parse the example files (fyi).

georgekucher, 2019-06-10:
@digitalentity I'm using servos in 2.1 as well, and no problems occur. BTW I even increased blackbox_rate_denom to 4. There was no such trouble even with 1.
But I'll try. Thanks.

metametaclass, 2019-06-10:
There is an 1024 character limit for header line in blackbox_decode and blackbox explorer:

https://github.com/betaflight/blackbox-log-viewer/blob/master/js/flightlog_parser.js#L447
`for (; stream.pos < lineStart + 1024 && stream.pos < stream.end; stream.pos++) {`

https://github.com/cleanflight/blackbox-tools/blob/master/src/parser.c#L356
`for (i = 0; i < 1024; i++) {`

and log with servos and debug_mode has header line with 1092 characters in it.

blackbox_decode with patched buffer size decodes log successfully.

digitalentity, 2019-06-10:
@metametaclass oh, nicely spotted!

digitalentity, 2019-06-10:
So, it's the decoder/viewer issue

stronnag, 2019-06-10:
For comparison, the 2.2.0 log attached doesn't have debug in it and it decodes OK in the **inav** `blackbox_decode`, 1036 bytes in the field header. The reported log at the head of the issue (with the same header) fails. Strange.
[LOG13773_2_2019-06-10_085901.TXT](https://github.com/iNavFlight/inav/files/3273368/LOG13773_2_2019-06-10_085901.TXT)

metametaclass, 2019-06-10:
> [LOG13773_2_2019-06-10_085901.TXT]
This log has 712 characters in header line.

stronnag, 2019-06-10:
Counting was never my string point ... Patching the inav decoder also makes the original log readable.

stronnag, 2019-06-10:
https://github.com/iNavFlight/blackbox-tools/pull/9

georgekucher, 2019-06-10:
Thank you all guys!

closed 2019-06-12 by digitalentity