iNavFlight/inav #4809: iNav firmware not working on MATEK F722-STD

biocage, 2019-06-09:
## Current Behavior
After flashing stable firmware 2.1.0, FC boots to a solid red LED, and Configurator is unable to connect. Serial port is present on host PC, but no traffic. Recovery requires powering the FC while the Boot button is held down, and then flashing a working Betaflight build.

## Steps to Reproduce
Flash iNav onto MATEK F722-STD using iNav Configurator 2.2.0RC2 (or RC1).

## Expected behavior
Expected to load firmware and then be able to connect via Configurator to begin configuration.

## Suggested solution(s)

digitalentity, 2019-06-09:
Can't reproduce.

Board: MATEKF722-STD
Firmware: MATEKF722
Version: 2.2.0 RC2

[image]

Double-check that you are flashing `MATEKF722` firmware, not the `MATEKF722SE` which is for MATEKF722-WING board.

stronnag, 2019-06-09:
and you flash with 'full chip erase'

biocage, 2019-06-09:
Yup, I did - on both counts. It looks like you are using different firmware than the latest "stable". Can you confirm that?

digitalentity, 2019-06-09:
@biocage 2.2.0-RC2 from the repository

biocage, 2019-06-09:
@digitalentity Ok, that's the same Configurator I'm using, but I saw the second line in your screenshot also says 2.2.0, and the firmware Configurator is offering me is 2.1.0 if I'm on the default stable channel. 

I poked "Show unstable releases" and flashed the unstable version you are using, and it does work. I'm going to update the text of this ticket to reflect the actual issue.

digitalentity, 2019-06-09:
Since the issue is fixed in 2.2.0, I'm closing the ticket.

closed 2019-06-09 by digitalentity