iNavFlight/inav #4817: set the last waypoint in mission planner to Landing instead of RTH. far 1 way missions

X-FOOT, 2019-06-10:
## Current Behavior

## Desired Behavior

## Suggested Solution

## Who does this impact? Who is this for?

## Additional context

peteoz, 2019-06-12:
it would be probably ideal if you can clarify few things as there are multiple permutations of what could happen when last WP is reached, depending on your other setup.
is this with or without radio signal ? is this with RTH after last WP enabled or disabled ?

there already is an option to disable RTH after last WP is reached, but that I believe only works when you still have a radio signal, and in that case model will not land but at least will not RTH and instead loiter around last WP.

I'm about to submit similar (but different) feature request for 1 way missions where I would actually prefer the model to loiter around last WP when the mission was flown without radio signal (with current implementation I believe the model will enter failsafe procedure when last WP is reached and radio signal has not been established, failsafe procedure would initiate RTH if that's your failsafe settings).

I'm doing 1 way missions whereby I send my model somewhere (I switch the radio OFF) and then I drive with my car to the location of the last WP, re-establish the radio link and land the model.
the problem with this approach is that if the model reaches the last WP before I get there, it'll be already travelling back to the launch site since my failsafe settings is RTH, so that's why I would prefer the model to loiter around last WP to give me chance to arrive there.

in your case you want option for the model to land at last WP, not sure how feasible this is since the terrain altitude at new location (last WP) may be different from your launch site altitude, iNav would need to know change in altitude for the new location (to land safely) and I'm not sure whether this can be accurately predicted from the GPS data. 
not sure what is your reasoning for the model to land at last WP but can imagine this to have some use in emergency situation where say you are stranded in remote area and want to send your model out to some known and reachable location for help with a message onboard.

breadoven, 2021-11-29:
Closed by:

https://github.com/iNavFlight/inav/pull/5485
https://github.com/iNavFlight/inav/pull/6822

closed 2021-11-29 by breadoven