iNavFlight/inav #4820: dynamic compass compensation

elepete, 2019-06-10:
## Current Behavior
Magnetometer must be placed distant to power cables / battery to avoid magnetic interference. But typically not all current interference can be eliminated.

## Desired Behavior
Current flow magnetometer offset could be compensated if calibrated vs. current.

## Suggested Solution
Needs in any case a board with current measurement.
A calibration could be done in a leveled ascent / descent without any yaw turn. In that condition magnetometer does (should) not change. A value table with mag x, y, z values vs. current could be stored to find the offset per ampere.
As the build geometry is fixed, the interference will be repeatable and the magnetometer read should be more stable.

## Who does this impact? Who is this for?
All users, that want to reduce current interference to magnetometer.
Could help with builds that have proximity between magnetometer and power cables / battery.

## Additional context

elepete, 2019-06-11:
I have looked to two flight logs that I have produced to see the blackbox data:
1. Compass mounted distant on mast:
Low interference even with 30 A delta to mag x, y, z data (kept the quad within 0.05g on x- and y-axis max. of 3° tilt)
2. Compass mounted close to battery (short mast):
Interference is visible in mag x, y, z data (again quad within 0.05g max on x- and y-axis)
From the log data I find it hard to make out an automatic calibration, there is too much sources of errors possible. Any idea is welcome.

An alternative idea: Strap the quad. stationary (as Joshua Bardwell does to calibrate the current sensor) and take the data from the blackbox. I will try that at a later point. That could than be manually calculated and entered as compensation.

hali9, 2019-07-29:
I've seen it already somewhere https://github.com/iNavFlight/inav/issues/93 https://github.com/iNavFlight/inav/pull/214

closed 2023-04-13 by DzikuVx