iNavFlight/inav #4822: Crossfire failsafe doesn't seem to work

teckel12, 2019-06-11:
INAV 2.2.0-RC2, Matek F722-SE & Crossfire. See attached log files and diff:

[crsf.zip](https://github.com/iNavFlight/inav/files/3274291/crsf.zip)

In both cases I set home position at while in position hold which should result in "SET-THR" failsafe as it's inside the `failsafe_min_distance`.

`LOG00001.TXT` I enable position hold, home reset and disconnect crossfire transmitter module, after about 15 seconds it cuts motors and drops (all of about 2 feet as it was a test).

`LOG00001.TXT` I enable position hold, home reset and throw RTH switch which does the expected "SET-THR" failsafe which is a quick landing at a throttle position of 1250.

I didn't try it beyond the `failsafe_min_distance` as it was really windy and I was in my side yard.  I played back the log files but I'm unskilled at knowing what to look for.

Jetrell, 2019-06-11:
@teckel12 Do you have the Crossfire FS set to - `Failsafe_Pos` or `Cut` ?
Sounds like an odd question. But from my tests it makes a difference.

If I use `Cut`, mine ignores `nav_min_rth_distance`.  But if I use `Faisafe_Pos`, it does not trigger FS until its greater than `nav_min_rth_distance`..

teckel12, 2019-06-11:
@Jetrell Sorry, yes, Crossfire FS is set to `cut`.

teckel12, 2019-06-11:
Looking at the logs, it shows `rxSignalReceived` being lost which triggers `failsafePhase`.  But it then just seems to wait for 20 seconds before cutting the motors.  The 20 seconds matches the delay for turning off the motors during `Land` failsafe (failsafe_off_delay = 200).

My guess...  Because I have position hold turned on at the time of the failsafe, that is taking precedence over what should be the correct failsafe procedure `Land`.  After 20 seconds, it cuts the motors as it should as that's the maximum time allowed to land before cutting motors.

I can't verify this is only happening when inside the `failsafe_min_distance`.  If outside `failsafe_min_distance` my RTH procedure would be to climb 100 feet before returning home.  And I'd rather not test a failsafe drop at 100 feet ;)

digitalentity, 2019-06-11:
Found a bug:

```
set failsafe_procedure = RTH
set failsafe_min_distance = 750
set failsafe_min_distance_procedure = SET-THR
```

The code checking whether we should disable any navigation modes (for `SET-THR`) only considers `failsafe_procedure` which is `RTH`. However that code doesn't consider that `failsafe_min_distance` overrides the procedure (in your case `SET-THR`). 

So, in your setup is quad is within `failsafe_min_distance` navigation is NOT disabled, however failsafe subsystem believes it's executing `SET-THR` and disarms after 20s.

teckel12, 2019-06-11:
Fixed with #4826

closed 2019-06-12 by digitalentity