iNavFlight/inav #4823: F405 wing. RTL fault

letruong6006, 2019-06-11:
I am using the latest INAV version for the F405 wing. in RTL mode. The plane flew in the circle above, I noticed that the plane flew about 5-6 loops above, the altitude dropped quite significantly and the radius was quite different from the settings, I installed the height of 120m and the radius of 50m . Throttle is about 30-40%, it flies quite fast, it is only accurate on the first flight above me, GPS is quite stable about 20 satellites, I use GPS BN280, my plane crashed into the bush because of the degree too low. Can someone help me?

Jetrell, 2019-06-11:
@letruong6006  At guess, with out seeing your Diff files.
Do you have - `nav_rth_home_altitude` set to a reasonable value?

If its set to `zero`, it will loiter at the `nav_rth_altitude`. But if it is set to something very low like 5mtrs. Then it will descend and most likely hit the ground.
Remember these CLI values are in Centimeters and not Meters.

letruong6006, 2019-06-12:
> @ letruong6006 Theo phỏng đoán, không cần nhìn thấy các tệp Diff của bạn. 
> Bạn có - `nav_rth_home_altitude`thiết lập một giá trị hợp lý?
> 
> Nếu nó được đặt thành `zero`, nó sẽ loiter tại `nav_rth_altitude`. Nhưng nếu nó được đặt ở mức rất thấp như 5 triệu. Sau đó, nó sẽ hạ xuống và rất có thể sẽ chạm đất. 
> Hãy nhớ các giá trị CLI này được tính bằng centimet chứ không phải là mét.

letruong6006, 2019-06-12:
> @ letruong6006 Theo phỏng đoán, không cần nhìn thấy các tệp Diff của bạn.
> Bạn có - `nav_rth_home_altitude`thiết lập một giá trị hợp lý?
> Nếu nó được đặt thành `zero`, nó sẽ loiter tại [...]

I setup nav_rth_home_altitude = 12000cm. 
The incident does not occur regularly

closed 2019-08-25 by stale[bot]