iNavFlight/inav #4828: nav_rth_home_altitude ignored on RTH after WP Mission

peteoz, 2019-06-12:
## Current Behavior
when RTH is enabled after last WP is reached in WP MIssion mode, it will not descend to nav_rth_home_altitude, instead it keeps loitering at nav_rth_altitude

## Steps to Reproduce
1. load mission with RTH enabled after last WP
2. execute and wait for mission to complete
3. watch RTH altitude
4. after reaching home point check that it doesn't descend to nav_rth_home_altitude

## Expected behavior
model should descend to nav_rth_home_altitude after reaching home point

## Suggested solution(s)
model should descend to nav_rth_home_altitude after reaching home point

## Additional context
# wp
#wp 11 valid
wp 0 1 -248767500 1338253000 10000 2250 0
wp 1 1 -248760000 1338250000 10000 2250 0
wp 2 1 -248716799 1338319100 10000 2250 0
wp 3 1 -248650900 1338312800 10000 2250 0
wp 4 1 -248604799 1338353100 10000 2250 0
wp 5 1 -248642100 1338368800 10000 2250 0
wp 6 1 -248647304 1338304348 10000 2250 0
wp 7 1 -248721095 1338322002 10000 2250 0
wp 8 1 -248797326 1338256159 10000 2250 0
wp 9 1 -248772767 1338243837 10000 2250 0
wp 10 4 0 0 0 0 165

set nav_rth_altitude = 10000
set nav_rth_home_altitude = 7000

---
# version
# INAV/MATEKF405SE 2.2.0 May 26 2019 / 19:56:35 (a713b63e4)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

hali9, 2019-06-13:
Not bug, new feature rather, because this is implement only for RTH in NAV_STATE_RTH_HOVER_ABOVE_HOME and not in WPRTH.
NAV_STATE_WAYPOINT_HOVER_ABOVE_HOME should be added.

Airwide, 2020-07-29:
I'm working on this feature.

breadoven, 2021-11-29:
Closed by https://github.com/iNavFlight/inav/pull/6524.

closed 2021-11-29 by breadoven