iNavFlight/inav #4829: please clarify 2 undocumented behaviours in WP Mission mode

peteoz, 2019-06-12:
1) nav_wp_safe_distance (allowed 0-65000)
- what does settings of "0" do ? does it mean there is no limit ?

2) when executing WP Mission with "failsafe_mission=OFF" and I don't enable RTH after last WP, then during the mission my radio loses signal or I switch the radio off, the Mission continues and completes. what happens if the radio link is not established by the time my model reached last WP ?  does it enter failsafe procedure or does it loiter around last WP ?

b14ckyy, 2019-06-12:
I'm interested in answers too as this could be part of one of my next manuals i plan to write.

teckel12, 2019-08-11:
@peteoz 

nav_wp_safe_distance = Don't allow arming if first waypoint is farther than configured safe distance

And a value of 0 is not unlimited.  A zero value wouldn't allot arming unless the first waypoint was exactly at the launch location.  So in reality, zero would almost never work, maybe it should mean no limit.

peteoz, 2019-08-12:
> nav_wp_safe_distance = Don't allow arming if first waypoint is farther than configured safe distance
> 
> And a value of 0 is not unlimited. A zero value wouldn't allot arming

@teckel12 thanks.

I find this limit of (max) 650m to be way too restrictive.
I just came back from my 2 months trip around Australian outback and I couldn't fly many pre-planned missions because of this limit.
I pre-planned all my missions at home (all my WP GPS coordinates) before leaving for the trip because once in the remote deserts you have no internet and mission planner only shows blank (white) map so you cannot modify or create new missions once you are on the road.
It happened several times that I could not reach my planned launch site because the terrain was not accessible, I tried to launch perhaps 1km away from my originally planned launch location and iNav won't allow you to even arm under that scenario.

I submitted feature request https://github.com/iNavFlight/inav/issues/4832
but no response so far.

teckel12, 2019-08-13:
@peteoz Can't you arm first then initiate the waypoint mission?  How I've always done a waypoint mission is load the mission, launch in angle mode, then hit my waypoint switch which initiates the mission.  I don't think this option even comes into play as it only restricts arming.

But maybe I'm doing it differently than everyone else.

peteoz, 2019-08-14:
@teckel12 
I load the mission from EEPROM with sticks command before arming, I don't think it's possible to use stick commands to load mission after the model was armed.
Then, once in the air I use the switch to activate WP Mission - that's when everything works normally.

But if load the mission when my location is more than 650m from the 1st WP then iNav will refuse to arm the model. 

> @peteoz Can't you arm first then initiate the waypoint mission? How I've always done a waypoint mission is load the mission, launch in angle mode, then hit my waypoint switch which initiates the [...]

teckel12, 2019-08-14:
@peteoz I just neglected to mention loading the mission from EEPROM.

I'll look into this...

digitalentity, 2019-08-14:
1 - is going to be fixed in scope of  #4832
2 - when you set `failsafe_mission=OFF` failsafe won't engage as long as WP mode is active, meaning INAV will complete the mission and loiter at last WP or perform an RTH (depending on the last waypoint type).

digitalentity, 2019-08-14:
PR in place #5013

teckel12, 2019-08-15:
@digitalentity I was playing with the code today and changed the feature to not disable arming, but instead just not allow the mission to start unless the model is inside the set safety zone.

I know this would change the feature from an arming limiter to a mission start limiter. But it seems to make more sense to allow the limit to be set to a realistic distance that you must fly inside before you can start the mission.

But, maybe it would just cause more confusion. I wrote the code and tested it today, but not sure if changing the way the feature works would be a good idea or not. 

And your simple fix does resolve the problem, but totally removes the safety aspect in the process. Thoughts?

peteoz, 2019-08-15:
> @digitalentity I was playing with the code today and changed the feature to not disable arming, but instead just not allow the mission to start unless the model is inside the set safety zone.

@teckel12 to me this makes perfect sense, it's actually what I suggested as one of the solutions when submitting feature request https://github.com/iNavFlight/inav/issues/4832
- this would allow me for example to fly my model manually as far as I can (ie. to an entry of a canyon) and then let the model fly through the canyon on its own with pre-planned WP mission
where WP1 would be at the entry just before I was to lose radio link
example of the terrain here : https://youtu.be/XNyx3n-bxEU

peteoz, 2019-08-15:
> 2 - when you set `failsafe_mission=OFF` failsafe won't engage as long as WP mode is active, meaning INAV will complete the mission and loiter at last WP or perform an RTH (depending on the last waypoint type).

@digitalentity thank you, that's fantastic news, that's what I was hoping for but wasn't sure if it was implemented that way.
scenario : I send out my model on WP Mission, switch my radio OFF and then drive with my car to a different location (ideally last WP of the mission), when I arrive at the rendezvous point I switch my radio back ON and take over the radio control again.
my obvious concern was what if I arrive too late and my model was already on its way back to launch site and I could not recover the radio signal, but if the model loiters around the last WP then there is no risk for this to happen

closed 2019-10-28 by stale[bot]