iNavFlight/inav #4832: sanity check for nav_wp_safe_distance should be only performed when activating the WP Mission, not when arming

peteoz, 2019-06-12:
## Current Behavior
when more than 650m (max value allowed) away from my 1st WP, I'm unable to arm, not sure if this is desirable behaviour because I may even decide not to fly the mission, and I want to be able to arm.

## Desired Behavior
sanity check for nav_wp_safe_distance should be only performed when activating the WP Mission, meaning once in the air, then and only then when I flip my switch to activate the WP Mission should the sanity check be performed and simply not execute the mission if my model is more then nav_wp_safe_distance away from 1st WP

## Suggested Solution
allow to arm the quad regardless of the 1st WP distance from the launch site, because it is not necessary to do the sanity/safety check unless the pilot decides to activate the mission once in the air. pilot may even decide not to activate the mission.
secondly the safety distance check should be measured between current position of the model and 1st WP (regardless of where the launch site is).
ex. if my nav_wp_safe_distance is 500m and my 1st WP is 2km North of my launch site, I could fly manually 1.5km North (to place my model 500m within reach of 1st WP) and then activate the mission, the mission should execute. however if my model was more than 500m away from 1st WP then the mission would not execute.

I also suggest to increase the maximum allowed value for nav_wp_safe_distance, currently 650m is way too restrictive and I found myself in situation where I could not arm and I could not relocate closer because the terrain did not allow me to.

## Who does this impact? Who is this for?
for pilots flying in unknown terrain, unknown location, for example in remote desert area. in these locations the mission planning is difficult because there is no internet. I plan the missions days ahead with no final knowledge of the terrain and not knowing local conditions, sometimes I find myself only 1km away from my intended launch site but I cannot get there due to terrain restriction, in this case I cannot arm my model.

## Additional context
there is no sanity check (restriction) when creating 2nd WP, it can be many kilometers away from 1st WP, so why is there such a severe restriction on 1st WP ?

when I forget to load my mission from EEPROM before arming and then try to execute mission once in the air, nothing happens because I'm trying to execute invalid mission.
in exactly same way should the sanity check for nav_wp_safe_distance be performed, only once in the air and only once the pilot tries to activate WP Mission, if the 1st WP is invalid then don't execute the mission.

peteoz, 2019-06-12:
this was submitted as a feature request, not a bug

peteoz, 2019-06-26:
further context.

sometimes I want to fly manual as far as I can (while I'm LOS) and only once entering a terrain where I know I would lose video and radio link (for example when entering canyon) I would like to switch to WP Mission only then (ie. assuming my 1st WP is at the entry of the canyon), but then my 1st WP might be several kilometres away from my actual launch site, under this scenario I wouldn't be even able to arm the model because my 1st WP is too far away. 

normally I want to fly model as far as I can in manual mode because still as a pilot I want to have the joy of controlling the plane and also I can get more usable footage out of manually controlled flight, but then once I reach a point in terrain where I know would lose my video and radio link I would still be happy for the model to continue autonomous mission.

I've been now travelling for 4 weeks in deserts and remote locations around Australia and came across many situations where this requested feature is desirable. for example on my recent flight where I planned to fly through a canyon in autonomous flight and I had to launch and land outside of the national park but the canyon entry was 1.5km away - normally I would've preferred to fly first part of the flight manually and activate WP Mission only at the entry to the canyon, video of the terrain and situation below, incl. the loss of video and radio signal for reference to also show that WP Mission in iNav 2.2 works pretty well even when there is failsafe :
https://youtu.be/XNyx3n-bxEU

digitalentity, 2019-08-14:
I'm going to add possibility to disable this check with `nav_wp_safe_distance = 0`

digitalentity, 2019-08-14:
PR in place #5013

teckel12, 2019-08-15:
To me, it seems more logical to do this sanity check when trying to start the mission, rather than when arming. This also allows one to fly inside the save distance and then start the mission.

danarrib, 2019-11-06:
I agree with @teckel12 , but I want to add that it would be nice if OSD shows a message on the Arming screen warning that WP1 is XXXX meters away... This would be a nice way to warn the pilot that the configured mission is wrong.

Also, when trying to engage NAV WP mode, it can blink the message "WP1 is too far away", so pilot can know why the NAV WP mode is not engaging.

breadoven, 2022-11-26:
Closed by https://github.com/iNavFlight/inav/pull/7107.

PR #5013 (2019-08-24): [NAV] Allow setting waypoint_safe_distance=0 to disable the check
Fixes #4832
closed 2022-11-26 by breadoven