iNavFlight/inav #4838: Compass calibration IST8310 does not work with inav 2.2 RC2

Ralfde, 2019-06-14:
The calibration does not show the 30 seconds any more.
Calibration does not work.
After it stopped the values are still 0 after save and reboot.
The compass seem to work. Just the graph is not shown correctly, because of the offset.

[image]

holoduke, 2019-06-27:
Same here. I actually tested various different GPS's (with compasses) on matek f411. Nothing really happens when pressing the calibrate compass button. no count down.

ninja-zx11, 2019-07-08:
Same problem here too.Configurator doesn't save values.Tries with HMC5883L compass on OmnibusF7V2.

Had to do compass calibration via transmitter stick combination.

gbluma, 2019-07-12:
Same results here. Compass is powered, enabled, and getting valid data. Shows up in graph. Only get zeros when calibrating. Setting i2c_speed to 100KHz did not change anything for me.

Compass is a HMC5883 on an Omnibus F4 v2.

digitalentity, 2019-07-12:
Please verify via CLI:
1. Run compass calibration
2. Go to CLI
3. Run "get magzero"

Verify magzero values.

gbluma, 2019-07-13:
Well. I can't explain it... but there are calibration values in there now. No re-calibration needed. I assume this means what I saw was an issue with INAV configurator not displaying data properly.

I'll erase the board and retest to see if the issue is reproducible. If there is an issue it would be nice to pinpoint it. I'll report back if anything seems worthwhile.

gbluma, 2019-07-13:
Retested. Issue is reproducible. I'll try to verify what I did to get it working.

1. Flashed with 2.2.1 firmware (from 2.2.0 configurator)
2. CLI "beeper -ON_USB" (as one does)
3. Saved and rebooted
4. Calibrated accelerometer
5. Saved and rebooted	
6. Calibrated compass
7. Saved and rebooted
8. Disconnected reloaded INAV (just in case of session issue)
9. Checked out calibration data (shows all zeros)

[image]

10. Checked CLI (get magzero shows zeros)

```
# get magzero
magzero_x = 0
Allowed range: -32768 - 32767

magzero_y = 0
Allowed range: -32768 - 32767

magzero_z = 0
Allowed range: -32768 - 32767
```

Again, Omnibus F4 v2 with HMC5883.

gbluma, 2019-07-13:
Update (sorry if this is getting chatty):

Using an older release of INAV Configurator (2.1.4) I can execute a compass/magnetometer calibration, but the current one (2.2.0) can't seem to do it.  2.1.4 works despite using the new 2.2.1 firmware, so it seems almost certainly on the Configurator side.

[image]

digitalentity, 2019-07-14:
Moving to Configurator repo. @DzikuVx canb you PTAL?