iNavFlight/inav #4842: inav altitude drop

DroneMichalovich, 2019-06-14:
Hey guys. I"m new here on the github forum can you guys help me with my 6" quad i have an altitude stable problem. When i arm the quad and just hover 2-5m above the ground
the OSD ALT value stays 0. Now if i land the quad it drops -5,-6,-8 and then when the quad is on the ground it come back to 0 again. Now if i fly a little then try RTH it starts to rth the quad start to climb BUT the numbers on the OSD ALT start to slowly get up in the meantime the quad starts to climb higher and higher above the set point because "It can't reach the setup point" until i disable the RTH.
Here are the logs to show you so that the best of you guys can help.
By the way i notice a strange baro behavior in black box program it see very jumpy to me.
It not go smooth like a normal baro do.
And of course AltHOLD act slightly jumpy. The other modes work just fine Air/Angle/PosHOLD.
What do you guys think that could be?

Michael.

[LOG00007.TXT](https://github.com/iNavFlight/inav/files/3292125/LOG00007.TXT)
[LOG00005.TXT](https://github.com/iNavFlight/inav/files/3292110/LOG00005.TXT)
[LOG00003.TXT](https://github.com/iNavFlight/inav/files/3288317/LOG00003.TXT)

teckel12, 2019-06-15:
@DroneMichalovich Do you have some foam covering the barometer?  Prop wash can cause barometer inaccuracy.

DroneMichalovich, 2019-06-15:
I'm using Matek F405-CTR. Now I do add some foam about 1.5x1.5x1cm(From runcam) under the board(On the baro)never have problems flying.
I setup the quad about 6 month's ago.
But about a week ago it starts I have noticed that when I'm in air the alt on OSD readings was floating around 0+-.
Really my only suggests is some how thru the foam some sand particle or dust get in to baro vent-hole. What do you think could it be ?
BTW the baro shows blue in the configurator like it working normal.

teckel12, 2019-06-15:
@DroneMichalovich It's not uncommon to have a few meters of inaccuracy. So at 10 feet showing 0 feet is quite common. Most of the time when I land the altitude shows  couple meters off from launch. That's about as good as the sensor is. Maybe your expectations are just too high? Without an optical flow sensor, expect altitude hold to move up/down several feet.

DroneMichalovich, 2019-06-15:
Listen I already write. I flight the quad like about 6 months. I'm didn't build the quad a week ago. Something happen with the baro sensor I'm pretty sure please check the log files for further info.
P.S I'm flight the quad for about half a year And everything work like a charm. Until a week ago that issue started.

digitalentity, 2019-06-15:
Sounds like a dead barometer

DroneMichalovich, 2019-06-16:
Tell me guys can a Matek f405-CTR barometer(bmp280) work just the baro it self without a magnetometer connected via i2c? Because when I try to power only the FC and set barometer to "auto"

giacomo892, 2019-06-17:
is perhaps the mag faulty and messes up with the I2C bus?

DroneMichalovich, 2019-06-17:
No the mag work just fine. I said when I connect the mag via i2c it work just fine and the baro shows "blue". But when I disconnect the mag the baro and the mag goes "red" if I try to turn them in tge configurator.
Is it a normal behavior or the baro must work even without the magnetometer?

Michael.

giacomo892, 2019-06-17:
the baro must work even without mag attached.

DroneMichalovich, 2019-06-17:
Even on Matek F405-CTR?

giacomo892, 2019-06-17:
Indeed

DroneMichalovich, 2019-06-17:
Maybe that is what making my altitude issue. But what i don't understand is why when I connect my Radiolonk ts100 GPS/Mag sensor baro status in CLI Via #Status shows me that the baro: BMP280= OK. If it is dead?

Michael.

giacomo892, 2019-06-17:
Disconnected the mag and enabled the baro only. If the baro doesn't work, then there's an issue.

DroneMichalovich, 2019-06-18:
Looks like my Baro are dead. I just replace my FC to new one.
Thank you all guys for helping me out.

Michael.

closed 2019-06-18 by DroneMichalovich