iNavFlight/inav #4843: External BMP280 detected as MS5611 on OMNIBUSF4 2.1.0

evranch, 2019-06-15:
When set to AUTO, it detects the BMP280 as a MS5611. It then gives it good health, but the reading does not change from zero. It will attempt to hold altitude using what I believe is the accelerometers alone, but this doesn't work well at all.

If I configure it as a BMP280 then it just reports it as having bad health.

I checked the target files and OMNIBUSF4 is supposed to look for the BMP280 on the external i2c bus. OMNIBUSF4PRO tries to use an internal bus, but I am not using the pro.

The BMP280 is a real one, ordered from Digikey on a Adafruit breakout board with built in level shifters and pullups. It is supplied with 3.3V from the FC. The I2C port is connected to the magnetometer of an M8N module which works properly. If the M8N is unplugged, there is no change in the behaviour.

stronnag, 2019-06-15:
Due to the weakness of the MS5611 PROM checksum there is always a small possibility that a baro device that is not working correctly will be misidentified (i.e. from  https://github.com/cleanflight/cleanflight/issues/1040). That your device does not work correctly when explicitly nominated indicates issues other than in the detection.

evranch, 2019-06-15:
Interesting. The fact that it is capable of triggering this bug implies that reads from the baro are succeeding, though. No I2C errors are being logged. Is there a way to get raw data from the I2C stream?

I will pull the baro module off the quad and attach it to an Arduino to check that it was not DOA. It may need something like stronger pullups, as well.

digitalentity, 2019-06-15:
You can specify `baro_hardware` manually

evranch, 2019-06-20:
Problem solved. The Adafruit board pulls SDO high by default for ease of use with SPI. This sets the I2C address to 0x77, and iNav is looking for it on 0x76. Pulling SDO down sets the address to 0x76 and it works perfectly. 

Changed the bug title to mention the Adafruit board in case anyone else ends up with this issue.

closed 2019-06-20 by evranch