iNavFlight/inav #4844: waypoint speed wont work

nkorls, 2019-06-15:
## Current Behavior
setup a mission with just one waypoint and rth  ,set altitude=15000cm, speed = 30m/s,the plane climb well and flying in a straight line and rth well，but the speed is slow,and throttle sounds low,i try to fly in manual mode,with half throttle ,the speed has exceeded 30m/s,the plane are capable of flying so fast but inav wont let it speed up
## Steps to Reproduce

1.setup a mission with one waypoint and rth,altitude=15000cm,speed=30m/s
2.take off and switch to wp mode
3.the plane began to climb
4.Speed cannot reach 30m/s,The throttle is very low.It sounds quiet.

## Expected behavior
After climbing, the speed reaches 30m/s  

## Suggested solution(s)
Increase the option of climbing while flying, or climbing and then flying flat 

## Additional context

---

- FC Board name and vendor: spracing f3
- INAV version string:2.1.0

digitalentity, 2019-06-15:
WAI. Speed control on a plane doesn't make much sense.

nkorls, 2019-06-15:
When I need it to fly faster, I want it to fly faster. Especially when there is a strong wind, it will theoretically fly backwards, with air speed and ground speed moving in different directions.

nkorls, 2019-06-15:
Now I can only indirectly control speed by setting cruise throttle.

b14ckyy, 2019-06-15:
And if you set a low speed and get tail wind, the plane will drop out of the sky. 
Inav always controls the throttle "blind" as the specific throttle value will maintain the same airspeed. 
There is a hardcoded minimum gps speed at 7m/s for automatic modes and rth. If the headwind is too strong, it will throttle up to maintain at least 7m/s. 

You can activate manual throttle override in cli to get faster during mission flights.

nkorls, 2019-06-16:
It seems that an airspeed meter is necessary.
i will try nav_fw_allow_manual_thr_increase,I still hope to set the minimum speed by myself,7m/s is too slow

peteoz, 2019-06-26:
if it helps I found that under WP Mission mode the actual speed that I set between different way points is actually ignored and instead iNav's Cruise Throttle settings (set nav_fw_cruise_thr) is used, in my case "1550" and certainly from my OSD footage I note that iNav tries to keep my throttle at 55% in WP Mission mode regardless of the speed settings. in general on my setup 55% throttle translates to +/- 80km/h depending on the wind direction.

digitalentity, 2019-06-27:
We may enforce waypoint GPS speed for fixed wings as "best effort" - INAV will not go below cruise throttle to avoid stall.

closed 2019-09-09 by stale[bot]

b14ckyy, 2019-09-10:
ping