iNavFlight/inav #4847: Additional flight mode: Minimum_Altitude

brandtaucher, 2019-06-16:
## Current Behavior
Currently the only mode to hold the altitude is altitude hold. However, this does not always give the desired security.

## Desired Behavior
It would be desirable that you can generally activate a minimum height that is not undercut.

## Suggested Solution
There should be a new mode "Minimum_Altitude_Mode", which acts as an additional mode on all other modes. He should never fall below a set height when activated via a switch. Once the minimum height is reached, it will continue to fly at the minimum altitude. You can specify a throttle setting that will help maintain this altitude (Minimum_Altitude_Throttle).

## Who does this impact? Who is this for?
For example, "Minimum_Altitude_Mode" is possible in Parrot Disco. But it is also useful in all multicopters. It helps with the following scenarios:
1. Beginners have more security
2. Aerobatics has an additional safety net
3. More security with fpv flight (perspective is sometimes misleading)
4. Altitude hold is improved. Due to variations in altitude, Altitude Hold may cause you to lose too much altitude.
5. etc

## Additional context
The text was generated using google translator. I hope you can understand everything. Please excuse.
Thank you!

nkorls, 2019-06-17:
and maximum distance limit
and rth after input has not changed for more than 10s

brandtaucher, 2019-06-17:
New name: Mode: phsyical limits

1. Minimum Altitude: in case - hold min. altitude 
2. Maximum Altitude: in case - hold max. Altitude
3. Maximum Range: in case: RTH
4. Maximum Failsafe: in case: RTH

mickey2011, 2019-06-20:
Maybe needs to be part of/merge with 4442

brandtaucher, 2019-06-21:
4442 is included in this. Physical Limits is an additional Feature for all modes.

brandtaucher, 2019-06-21:
4442 is included in this. Physical limits is an additional function to all modes.

colert, 2019-06-22:
I don't think there should be a new mode. Instead of that, there could be new optional settings for the altitude hold mode as you stated.

brandtaucher, 2019-07-02:
Nice that you are dealing with it. But you did not understand what it is about. If minimum altitude is enabled, you will not be able to land, etc. Therefore, it is necessary that there is a status that you can deactivate again.

goodshort, 2021-04-06:
Interesting request, I'm following! I recently crashed into bramble trees and such feature would have save my legs. I could see it used as well while flying over water.

closed 2024-02-02 by DzikuVx