iNavFlight/inav #4848: Weird disarm flags

tangettyplayz141, 2019-06-16:
I know theres alot of fourms about people not being able to arm inav but I haven't seen any with similar disarm flags
I already did all of the normal checks to get it to arm everythings showing green so I type status in cli while the systems powered and I get the flag "0x140000" if I flip the arm switch and type the same thing I get "0x184000" then after I disarm I get "0x180000". I am new to inav so sorry if it's an obvious problem, currently inav is running on an old spracing f3 setup for a wing

stronnag, 2019-06-16:
The wiki entry https://github.com/iNavFlight/inav/wiki/"Something"-is-disabled----Reasons may assist you in decoding the arming mask values.

tangettyplayz141, 2019-06-16:
> The wiki entry [https://github.com/iNavFlight/inav/wiki/"Something"-is-disabled----Reasons](https://github.com/iNavFlight/inav/wiki/%22Something%22-is-disabled----Reasons) may assist you in decoding the arming mask values.

yea thanks but i already took a look at it and my codes dont match any of the ones listed and if you actually have to "decode" the flags i would have no idea how to do that

digitalentity, 2019-06-16:
0x140000 = CLI + RX
0x184000 = CLI + THR + ARMSW
0x180000 = CLI + THR

Hope that helps you figure out that your throttle is not at zero (probably inverted).
Also, see https://www.youtube.com/watch?v=fVJYodLimD8, especially starting at 3:35

stronnag, 2019-06-16:
> yea thanks but i already took a look at it and my codes dont match any of the ones listed 

They do .... configuration and / or user errors:

0x140000 = 00100000 + 00040000 (CLI active, RX not detected)
0x180000 = 00100000 + 00080000 (CLI active, throttle not at minimum)
0x184000 = 00100000 + 00080000 + 00004000 (CLI active, throttle not at minimum, arm engaged)

You can discount the CLI flag (as you're in the CLI deliberately), you need to address the RX and throttle cases.

tangettyplayz141, 2019-06-16:
ok thanks for replying so fast as i said im new to i nav so i had no idea that the one number corresponded to multiple flags. appreciate the help

closed 2019-06-16 by tangettyplayz141