iNavFlight/inav #4855: SA 2.1 not working in iNAV 2.2 final

Nihilistic, 2019-06-19:
## Current Behavior

## Steps to Reproduce

1.
2.
3.
4.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:Matek F405-CTR
- INAV version string:
SA 2.1 not working with TBS Unify Pro32 HV MMCX.

teckel12, 2019-07-02:
Something that may help with SA 2.1:

https://github.com/betaflight/betaflight/pull/8276

crybeFPV, 2019-07-07:
Same here in 2.2.1 on Matek F722-STD withTBS Unify Pro32. Flashed forth and back with Betaflight 4.0.4 where everything is working fine. 
Please merge the current Betaflight Code for Smart Audio. Thanks for all your work in the inav project @devs

teckel12, 2019-07-07:
@crybeFPV Looking at the Betaflight changes, they're fairly drastic for SA 2.1 support. They changed a bunch of things so it's not at all as simple as merging some code into INAV. There will be a significant amount or re-work involved or it will need to be re-written to work with the INAV code.

The other issue is that no core INAV developers have a SA 2.1 VTx to test with. TBS is going to supply a few SA 2.1 VTxs for development/testing, but there's no promises on when this will fit into anyone's interest schedule.

Finally, I've heard that some are still having issues with SA 2.1 in Betaflight. So I'm not totally sure the Betaflight code can be used as-is (again, test equipment is needed to confirm or deny this).

Basically, it's known that it doesn't work correctly in INAV at this time. Also, pull requests with the Betaflight SA 2.1 code is encouraged.

Jetrell, 2019-07-09:
Just thought i'd say, that I'm interested in seeing SA 2.1 working at some later time, as well. 

@teckel12 I'm finding that the 2.1 BF development SA code is working fine.
 I think most of the problem... Is that users aren't loading the correct table for their VTX... or aren't loading a table at all.

teckel12, 2019-07-09:
@Jetrell That appears to be correct, that the current problems in the latest release of Betaflight are that people are not setting up their VTx correctly.  That's also exactly where the complication is with porting the Betaflight code into INAV.  A lot was changed in Betaflight to create the VTx tables, and going that route may not even be an option for INAV.

Jetrell, 2019-07-09:
To be honest. I'm not totally sure that the `tables` are that important anyway. 
It appears to me, that the main reason they have been implemented. Is to keep users compliant with there local  frequency laws and power levels.
We have got by just fine, selecting the correct frequencies and power levels in the past. Perhaps the port from BF could be simplified much more, by not including the table function at all.

teckel12, 2019-07-10:
@Jetrell I believe it's part of the same PR, or at least based on it. In any case, it adds to the complexity.

I would like to see some type of customization of the VTx power levels.  It's kinda a hot mess right now with every VTx manufacture having a different number of steps and different mW levels. It would be nice if the user could select the number of steps and each of the mW levels as there's a lot of confusion with the current system.

We are, however, getting ahead of ourselves. I just talked with Trappy again today about getting SA 2.1 samples to begin testing with. I don't see any progress being made till there's samples in developer's hands as obviously trying to implement SA 2.1 support without the hardware wasn't successful.

Jetrell, 2019-07-10:
Thanks for talking to Trappy about the matter. Hopefully he can forward a few to the guys here :+1:

kreidler, 2019-08-07:
I just want to queue into this issue having the same problems with my first wing on a Matek F405-WING and the new TBS UNIFY PRO32 HV. 
We are running some of the smaller Nano Pro32 version on BF4.0 (SA2.1) without any problems.
BF is still using numbers for the power levels (1 to 4). Nevertheless SA2.1 uses dBm in the responses which could be used to get the "real" mW displayed.

putimir, 2019-08-14:
Similar situation here: Matek F411-WING, 2.2.1 final, SA on ST1 soft, TBS UNIFY PRO32 HV, 1.09 FW.

I do however get the OSD item populated, also it seems everything is ok in the OSD/sticks menu, but it does not respond to commands....

DzikuVx, 2020-06-11:
SA2.1 works with INAv 2.5

closed 2020-06-11 by DzikuVx