iNavFlight/inav #4856: can't calibrate BN-880's magnometer (HMC5883)

goodnerdboy, 2019-06-19:
## Current Behavior
Mag's icon stays red and the calibration doesn't feed the calibration's parameters

## Steps to Reproduce
1. Install BN-880 on Foxeer F722dual
2. Setup ports, configuration, etc
3. As the auto mode of mag doesn't work, force it to HMC5883
4. try to calibrate

## Expected behavior
A correct calibration happens.
For information, if it can help, these steps work with Betaflight 4.0.3. At least it shows it is not a faulty mag.

## Suggested solution(s)
Unfortunatly i have no clue sorry.

## Additional context
As i wrote above, the steps work in Betaflight 4.0.3 and the readings from the mag look correct. This probably means the mag is not faulty and the connections are correct.

https://pastebin.com/zmkBp7Rr
---

- FC Board name and vendor:
# INAV/FOXEERF722DUAL 2.2.0 Jun 18 2019 / 14:57:00 (5e5551bd3)
- INAV version string:
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

digitalentity, 2019-06-19:
Please provide a wiring diagram showing how compass is connected

goodnerdboy, 2019-06-19:
Hi.
I followed the wiring diagram provided by Foxeer.
Here it is :

[image]

digitalentity, 2019-06-19:
Thanks. Will investigate. Fortunately I have an FC at hand

goodnerdboy, 2019-06-19:
Many thanks!
Tell me if I can be of any help.

goodnerdboy, 2019-06-25:
Hi @digitalentity 
Did you have anytime to investigate this ?
Is there any way I can help?

goodnerdboy, 2019-06-25:
So this target declares the Baro on i2c 1 and Mag on i2c 2.
Unfortunately only i2c 1 is associated to pins, so I guess this target allows to connect an external barometer but not a compass.

goodnerdboy, 2019-06-25:
Should it be modified to match the manufacturer’s wiring diagram or is it purposely like that?

goodnerdboy, 2019-07-03:
Hello.
Any update on this one?

digitalentity, 2019-07-03:
Please test this hex

[inav_2.2.1_FOXEERF722DUAL.zip](https://github.com/iNavFlight/inav/files/3356764/inav_2.2.1_FOXEERF722DUAL.zip)

goodnerdboy, 2019-07-08:
It seems to work so far, I was able to set it up and calibrate it.
Thanks :)

digitalentity, 2019-07-08:
Thanks for testing. Fixed by #4908

closed 2019-07-08 by digitalentity