iNavFlight/inav #4862: 2.2.0 Smartaudio 2.1 not working via softserial on Matek F411

jasonsmit4, 2019-06-22:
## Current Behavior

## Steps to Reproduce

1. Install iNav 2.2.0 on Matek F411
2. Wire TBS Pro32 HV to ST1 pad
3. Enable Softserial in settings
4. Configure smartaudio on ports tab
5. Does not work, dashes appear in OSD where band/channel/power should be.
6. SA control works correctly with the exact same settings/config/wiring with an older TBS VTX with SA 2.0.

## Expected behavior

Smartaudio control via OSD/stick menus and LUA script should work

## Suggested solution(s)

Investigate compatibility with SA2.1 and any potential issues specific to softserial.
## Additional context

https://pastebin.com/hfsT0uV3
---

- FC Board name and vendor:
Matek F411
- INAV version string:
# INAV/MATEKF411 2.2.0 Jun 18 2019 / 14:54:48 (5e5551bd3)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

giacomo892, 2019-06-22:
Duplicate of #4855

Closing

closed 2019-06-22 by giacomo892

digitalentity, 2019-06-22:
Thanks, moved the issue details to #4855