iNavFlight/inav #4863: MATEKF411 target doesn't work for F411-based Eachine Tyro79 quad, please fix or add separate target

georgie3, 2019-06-22:
## Current Behavior

MATEKF411 target doesn't work on F411-based Eachine Tyro79.  After flashing iNav the board boots and the LED blinks slowly then rapidly over and over, all communication fails. Betaflight MATEKF411 target works fine. 
## Desired Behavior

Please add support for the custom STM32F411-based Eachine Tyro79 FC. 
## Suggested Solution

Could just need a tweak to the MATEKF411 target, or could be a separate target. I'm a developer and could help, but I'd have to painfully compare the iNav and Betaflight F411 implementations and try things until it boots properly. I can help testing.
## Who does this impact? Who is this for?

Tyro79 flyers who find Betaflight boring and love iNav :)
## Additional context

digitalentity, 2019-06-22:
Suspect that they use the different gyro chip. Can you provide Betaflight's `status` command output?

georgie3, 2019-06-22:
Thanks for the quick response... yes it's definitely a different gyro/acc chip, ICM20602. Here's Betaflight 4.0.3 status output:

MCU F411 Clock=96MHz (PLLP-HSE), Vref=3.26V, Core temp=31degC
Stack size: 2048, Stack address: 0x2001fff0
Config size: 3000, Max available config: 16384
GYRO=ICM20602, ACC=ICM20602
System Uptime: 41 seconds, Current Time: 2019-06-22T19:00:28.205+00:00
CPU:14%, cycle time: 126, GYRO rate: 7936, RX rate: 33, System rate: 9
Voltage: 1 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 3
Arming disable flags: RXLOSS ANGLE CLI MSP

digitalentity, 2019-06-22:
Try this firmware (it includes drivers for MPU6500/ICM20602 gyro and should work if it's connected the same way):

[inav_2.2.0_MATEKF411.hex.zip](https://github.com/iNavFlight/inav/files/3317206/inav_2.2.0_MATEKF411.hex.zip)

georgie3, 2019-06-23:
Awesome, this works. Gyro and acc behave normally in configurator sensor graphs and 3D OpenGL view. I can't flight test yet because I'm still waiting for a receiver that fits in my Tyro79, but all looks good... motors run in test mode.

digitalentity, 2019-06-23:
Good, closing this then

closed 2019-06-23 by digitalentity