iNavFlight/inav #4864: VTX Power switching in flight

craigywsm, 2019-06-22:
Hi All,
Please could someone tell me how to change VTX power in flight assigned to a switch?
I have tried using the adjustments page with no luck
Currently using INAV 2.1 with the Matek F722 wing board with the Matek VTX-HV with IRC TRAMP

teckel12, 2019-06-22:
@craigywsm I don't think you can from a switch, you can from the OSD menu via stick commands. See here:

https://github.com/iNavFlight/inav/blob/development/docs/Controls.md

craigywsm, 2019-06-22:
Thanks for your info teckel. 
Just done some research and found it on the latest 2.2.0 release so i will need to upgrade and re test with the adjustments the setting must have been there for future implementation 
https://github.com/iNavFlight/inav/releases
[image]
Could really do with some documentation on how this works if at all possible.
Happy to lend a hand writing it if someone can point me in the right direction.
This might need further development as may need to store the power levels as each vtx has different power levels.

teckel12, 2019-06-22:
@craigywsm That's a new feature I wasn't aware was added! =)  I assume it would be done via the Modes tab like all other switch-based controls are configured.

The state of power levels is a mess, There will be no way that INAV can keep up with the different models that are released every week with totally different power levels.  Best to just set it to 1-5 or whatever and call it a day.

L4ky, 2019-06-24:
> @craigywsm That's a new feature I wasn't aware was added! =) I assume it would be done via the Modes tab like all other switch-based controls are configured.
>
> The state of power levels is a mess, [...]

No, this function is handled by Adjustments tab.
Just choose your activation switch ( it could be the arm switch ), VTX power level in the dropdown menu, and the command switch which will be the one that will increase/decrease the value.
My advice is to use a 3-pos switch or a stepped-potentiometer ( i don't know the exact name ).

Remember: you don't have to keep your switch up or down too much or you will make more than one step.

Try that on the ground to learn its effectiveness.

P.S. Add VTX power on OSD to see current power level

CapnBry, 2019-06-24:
There's a great video on setting it up and using it on youtube [iNav 2.2.0 VTX power change in flight](https://www.youtube.com/watch?v=Vd6EBr4QYa8) from NoAgenda FPV. I also wasn't aware of this feature until he posted the video. Skip to about 1:30 in the video to get to setting it up.

P-I-Engineer, 2019-06-24:
i'd prefer this was in modes as well, or is there a way to do position based values instead of incrementing?

if it was modes, i would use a 3 pos switch and set my max values in opentx to set the power level to the switch position.

L4ky, 2019-06-24:
Actually you can not assign values based on position.

You can only increase or decrease current value.

craigywsm, 2019-06-24:
CapnBry many thanks great video will test this on my setup.
P-I-Engineer and L4ky. Yes it will be like the PID adjustment theory.
I see there could be an issue here as all vtx's have different power levels might be worth adding settings for the ones the VTX can support. Not sure on how the Tramp / Smartaudio with handle the changes
e.g 25mW , 100mw, 500mw
or 25mw, 200mw, 800mw???
With a county override

P-I-Engineer, 2019-06-24:
my workaround for time based incrementing is going to be a "duration" logical switch. So that moving switch up will increase by one, then back to middle, then back to up to increase again. Same for decrease. That way you don't have to time it with your fingers.

using telemetry you could have it auto increase also with logical switches, but that's not the smartest way to do it, since it's a blind effort.

L4ky, 2019-06-24:
> CapnBry many thanks great video will test this on my setup.
> P-I-Engineer and L4ky. Yes it will be like the PID adjustment theory.
> I see there could be an issue here as all vtx's have different power [...]

The logic behind all this is "power levels" which are (usually) number from 1 to 5. 25/100/200/500/800 are just labels but they don't exists in the protocols..

At the moment most vtx have 4 or 5 different power levels so not a big issue there.
The problem will be in the future if power level count will increase.

Another user is working on this.. i think you'll see something soon

P-I-Engineer, 2019-06-24:
here's my opentx work around for 1 position increment.
[image]
[image]

noagendafpv, 2019-06-24:
> There's a great video on setting it up and using it on youtube [iNav 2.2.0 VTX power change in flight](https://www.youtube.com/watch?v=Vd6EBr4QYa8) from NoAgenda FPV. I also wasn't aware of this feature until he posted the video. Skip to about 1:30 in the video to get to setting it up.

Thanks for sharing, I just stumbled on this and was going to post the link, but you beat me to it.

Now I have to try what PI Engineer did for Auto switching.

teckel12, 2019-06-24:
Now we just need VTx power levels to be set automatically based on distance from home =)

P-I-Engineer, 2019-06-24:
here's simpler mix. I set it to the throttle trim as well.

[image]
[image]

Jetrell, 2019-07-14:
Hi guys. 
To all you that have tried this VTX power level change.
I played around a bit with Companion, while i'm away. But have no way to test it for some weeks.

This OpenTX combo of Logic switches and Special functions. Seems to work in the simulator. 
I was wondering if one of you guys with an OpenTX radio could try it out and report back on how it works. 
It should perform more like a mode function. It has 4 positions (disarmed 25mw, Pos1, Pos2 and Pos3) I based it on a Unify 25, 200, 500 and 800.

It uses the arming switch to return to VTX to 25mW, when disarmed, and holds it at the lowest position.  

Then when the armed, it looks at the location of the 3 Position power selector switch. And from that it selects the appropriate timer required to place the VTX power level at the selected switch position. 
It will also speak the power levels when the power level switch or disarming is toggled.

The timers will need to be altered to get it to work correctly. I just based the timers on 0.6seconds, to administer a single change in level.  1.2seconds for two changes. 1.8seconds for three changes.

The zip also contains power level audio files.

[image]

[image]

[VTX power change - OpenTX.zip](https://github.com/iNavFlight/inav/files/3389935/VTX.power.change.-.OpenTX.zip)

teckel12, 2019-07-14:
@Jetrell I've been playing with something similar to automatically set the power level based on distance from home.  It's a little messy because you can't directly select a power level.

The plan is to eventually do this automatically right from within Lua Telemetry without any logical switches or special functions.  Also, this will allow selecting an exact power level instead of holding a logical switch for a certain amount of time (which is prone to not always working).  The thought is something like power level 2 at 100 feet, power level 3 at 1000 feet, etc.  Maybe even raise the power level when facing towards the pilot instead of away from the pilot (where the antenna is sometimes blocked by the model).

This could also be done in INAV, which is probably a better option.

Jetrell, 2019-08-13:
Hi Guys. I finally got around to testing this OpenTX logic switch and Special functions mix for the VTX power control.
It uses a 3Pos switch to set 200mW, 500mW and 1000mW and the ARM switch to drop back to 25mW when DisARMED. 
When ARMED, it will go to what ever power level you leave the power switch in. 
I have given it a good workout, and it doesn't get muddled up. The only thing that needs to be avoided, is flicking through the 3Pos power switch too quickly. i.e. don't flick from Pos1 to Pos3 in a rush. Make sure Pos2 is registered for 0.3 seconds.

This timing was set using a Unify Pro. I haven't tested Tramp. 
Make sure that - `set vtx_low_power_disarm = OFF` in iNav. This is because the timing in iNav, for re-arming isn't as precise as the mixer requires.

[image]

[image]

[image]

[mW audio.zip](https://github.com/iNavFlight/inav/files/3499176/mW.audio.zip)

DzikuVx, 2019-10-08:
Power switching in flight is possible now with 2.2 with adjustments, and in 2.3 will be a part of Global Functions. Closing. Thanks

closed 2019-10-08 by DzikuVx

RomanLut, 2020-12-23:
There is a video describing how to do it in 2.6.0:
https://www.youtube.com/watch?v=g3N7e_SHSwM