iNavFlight/inav #4869: Autotune might be storing Dgain instead of FFgains

DzikuVx, 2019-06-23:
## Current Behavior
INAV 2.2 Autotune might be storing Dgain instead of FFgains. Not 100% confirmed since it happened only on one of the axises

## Steps to Reproduce
1. Run Autotune
2. Check gains

## Expected behavior
Store FFgains as FFgains and Dgains should be 0 on fixed wings

digitalentity, 2019-06-23:
Confirmed via reading the code. Fix pending

CapnBry, 2019-06-24:
The D/FF values are also not being logged to Blackbox (#4839), and their setting isn't included in the blackbox info header, nor are they showing in the OSD since the extra field was added to the PID struct. There were a bunch of small things broken by adding this late change that need to be found and cleaned up now.

PR #4870 (2019-06-24): [AUTOTUNE] Fix incorrect logic modifying D-gain instead of FF-gain
Fix #4869, bug introduced by 9d7b27c3450ae567757d58136d07a1d2210e034f (only in 2.2.0). 

TL;DR Autotune was updating D-gains instead of FF-gains because we didn't consider it in 9d7b27c3450ae567757d58136d07a1d2210e034f
closed 2019-06-27 by digitalentity