iNavFlight/inav #4872: Raw input mixed with stabilized input ?

Bogdanu2018, 2019-06-24:
Hello 
So I have this issue, if I set my funjet with Stabilized Roll and Pitch in Mixing panel I'm not able to use the radio trim during flight in Manual mode.
I actually want the INAV only for OSD and failsafe RTH. I don't care about horizon or other modes.
I want only raw input with radio trim functional but still have the RTH . Is it possible ?
Is it possible to activate Stabilize Roll and Pitch only in RTH ?

Thank you

Dronek, 2019-06-27:
I suggest using the "servo autotrim" feature to trim your aircraft and then fly in "manual mode" for the raw steering experience. There is basically no need to trim manually in inav.

Alternatively you could set up some conditions in your radio with trims only active when the mode switch is in the position for "manual mode" in inav. And trims disabled or set to zero when the mode switch is set to RTH.

DzikuVx, 2019-06-27:
RTH works only with stabilized mixer inputs. Use Autotrim and then Manual mode if you want.

closed 2019-06-27 by DzikuVx