iNavFlight/inav #4881: Support for MS5525 based Digital Airspeed Sensors

Rusty105, 2019-06-25:
## Current Behavior
iNav currently uses MS4425 based Airspeed Sensors

## Desired Behavior
Would like to also have the option of MS5525 Digital Airspeed sensors

## Suggested Solution
Add option in code for the MS5525 sensors

## Who does this impact? Who is this for?
Airplane  users

The MS5525 is currently about 1/2 price of the MS4425.

digitalentity, 2019-06-27:
Are those sensors available for sale?
Also, consider https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md

Rusty105, 2019-06-28:
They are for sale, mRobotics sells them, when in stock. I just thought it might be a more affordable option.
https://store.mrobotics.io/mRo-Next-Gen-MS5525-Airspeed-Sensor-NEW-p/mro-ms5525v2-mr.htm

digitalentity, 2019-06-29:
$40 shipped and including the Pitot tube on AliExpress for MS4525 (already supported in INAV) http://s.click.aliexpress.com/e/bJpnhHxa
$60 + shipping for MS5525 sensor - doesn't look a more affordable option to me :smile:

Rusty105, 2019-06-30:
Was referring to the bare  sensor price, ~$22 for the MS5525 vs ~$38 for the MS4525. Not sure if it was a recent price increase for the MS4525 or not. Even in quantity the 5525 is about 1/2 the price. Just something to think about. If the DEVS don't want to, then so be it.

MS5525, Mouser and Digikey
https://www.mouser.com/ProductDetail/Measurement-Specialties/5525DSO-DB001DS?qs=%2Fha2pyFaduiVSgcyvwWHG3Xdem4Ijh4cI39uGWyFvzCgLliFXzSZLd%2FgWip9mXgL

https://www.digikey.com/product-detail/en/te-connectivity-measurement-specialties/5525DSO-DB001DS/223-1707-5-ND/5824908

MS4525, Mouser and Digikey
https://www.mouser.com/ProductDetail/Measurement-Specialties/4525DO-DS5AI001DP?qs=%2Fha2pyFaduhNZ6rksjza2aI6ciE0CSTf412WCq3SAwM%3D

https://www.digikey.com/products/en/sensors-transducers/pressure-sensors-transducers/512?k=ms4525&k=&pkeyword=ms4525&sv=0&s=49330&pv218=4&pv41=146&sf=0&quantity=&ColumnSort=0&page=1&pageSize=25

digitalentity, 2019-06-30:
Bare sensor price only matters to manufacturers. I don't think we should add support for any sensor that's out there, but rather only ones that are widely available to the hobbyists. 

I suggest we keep this issue open and revisit it in a while to check if it's more widely available.

leeph, 2019-09-12:
I would love to see support for the EagleTree sensor. There was a PR open for it a while back but I think development has stalled.

FlowRanger, 2020-05-15:
Could we revisit this? I'd love to see this sensor supported in iNav. In my testing it's more accurate especially at low speeds and it comes with temperature compensation. I'd be happy to purchase the 2 sensors required per the new hardware policy if that helps.

closed 2024-02-02 by DzikuVx