iNavFlight/inav #4886: Analog airspee dsensor on adc #3 matek f405wing and f722 wing

wx4cb, 2019-06-27:
## Current Behavior
Currently there is no way to use an analog airspeed sensor (such as the one in the F35 lightning on a matek f405 or F722 wing.

## Desired Behavior
Allow the use of the RSSI pin ADC for analog airspeed

## Suggested Solution
modify the target.h file to change 

IE Change this:
#define RSSI_ADC_CHANNEL            ADC_CHN_3

to This:
#define AIRSPEED_ADC_CHANNEL            ADC_CHN_3

Additionally, be able to "DEFINE" in the CLI whether you are wanting rssi or airspeed on that channel.

## Who does this impact? Who is this for?
Anyone who would like an analog airspeed sensor on a matek f405 wing or F722 wing that are moving over from the F35 lightning or just want an airspeed sensor. Obviousle if you actually use the RSSI pin FOR RSSI this wouldn't work, but for others that are using either crossfire with native RSSI/LQ or are sending it back on a transmitter channel, having this option would be great.

DzikuVx, 2019-06-27:
On matek f405-wing voltage divider is required and then any ADC channel can be mapped to pitor
```
get adc_channel and note channel assigned to RSSI pin. In my case it was channel 3
set rssi_adc_channel=0 – disable RSSI ADC
set airspeed_adc_channel=3 – map ADC channel 3 to airspeed sensor
set pitot_hardware=ADC – enable analog pitot tube
save
```

on F722-WING there is ADC4 labeled `AirS` and it already has a voltage divider  built in. So:
`set airspeed_adc_channel=4`

closed 2019-06-27 by DzikuVx

wx4cb, 2019-06-27:
OK cool.. that's great pawel. looks like it's going to be the 722-wing then in this bird :D I was hopeing for that as I have more of those and really like the 2 camera option :D

billyd60, 2020-04-14:
I am using a 722-wing and INAV
In reading the documentation for this board, the MATEK manual states that a 1:1 divider resistance is built in, lol So this may be a typo, however I am getting 234km/hr on the bench when sitting still. So perhaps it is not a typo! Has anyone tried this? Perhaps I should just connect the 1/2 solder connector on the sensor itself, since it doesn't seem as though the wing is dividing the voltage at all. Thoughts?