iNavFlight/inav #4890: BB error with Matek F722-SE

xabc1, 2019-06-30:
Having this error when opening the logs.

[[image]
](url)

Inav Firmware: Inav 2.2
FC: MatekF722SE
BB: Inav BB 2.3

Debug_mode GYRO is enabled.
[LOG00011_MatekF722SE.TXT](https://github.com/iNavFlight/inav/files/3342542/LOG00011_MatekF722SE.TXT)

I used another FC, this time Kakute F7 with the same setup and Debug_mode GYRO is also enabled with no problem with opening the logs.
[LOG00007_KakuteF7.TXT](https://github.com/iNavFlight/inav/files/3342547/LOG00007_KakuteF7.TXT)

stronnag, 2019-06-30:
This is not  a firmware issues, it's a viewer issue, already fixed in the iNav blackbox tools https://github.com/iNavFlight/blackbox-tools/pull/9

A separate issue already exists for the iNav BB viewer https://github.com/iNavFlight/blackbox-log-viewer/issues/36

stronnag, 2019-06-30:
fixed by https://github.com/iNavFlight/blackbox-log-viewer/pull/37

closed 2019-06-30 by stronnag