iNavFlight/inav #4891: Accelerometer calibration leads to config reset on Omnibus F4 Pro V3 board (INAV 2.2)

etulupov, 2019-07-01:
## Current Behavior
Accelerometer calibration leads to config reset.

## Steps to Reproduce
1. Start accelerometer calibration process
2. Perform several calibration steps.

Calibration is not completed. Observed FC restart and after boot up configuration is empty. It's restored to the factory defaults.

## Expected behavior
Calibration should be successful and config should be persisted.

## Additional context
INAV 2.1 works as expected.

rrk1, 2019-07-01:
I can confirm that was happened once with 2.1 too. INAV 2.1 + configurator 2.1.4

Exactly, config was reset to default after acc calibration attempt initiated in configurator page. May be I have tried to calibrate different sides not in default sequence. Target was DYSF4PROV2 (or maybe MATEKF405)

teckel12, 2019-07-01:
I believe you need to do it in the correct sequence or it won't work.

sam2b, 2019-07-02:
@teckel12 and the config reset?  That is the actual problem stated.

I had the exact same problem (a config reset) during the accelerometer calibration with 2.2.0 on a new Matek F722SE flight controller.

rrk1, 2019-07-02:
Additional problem with acc calibration sequence (possibly worth opening another issue) is that steps 3 and 5 are hard to do properly on asymmetrical hex frame like RD290 because it does not have a good axial reference. Wiki says 'the algorithm does not care about exact positions as long as they are close to 90 degree apart' but it was never stated how close to 90 degrees is okay,

giacomo892, 2019-07-02:
The configuration reset happens when you start the procedure but you don't finish it. 
That might have to be fixed indeed.

adriancodre, 2020-04-28:
yes, it happened to me twice on a Matek405-CTR. I did not noticed why, but now it makes sense

giacomo892, 2020-04-28:
fixed in #5527

closed 2020-04-28 by giacomo892