iNavFlight/inav #4892: Galileo not working in 2.2 final?

silenec, 2019-07-01:
When I set gps_ublox_use_galileo = ON and reboot FC, the "Drop ratio" in configurator goes to 99% and FC freezes after a few seconds from reboot.
With gps_ublox_use_galileo = OFF everything works in 2.2
gps_ublox_use_galileo = ON works with iNav 2.1

HW = Omnibus F3 and Beitian BN-180 GPS

Is it just me with this problem?

stronnag, 2019-07-01:
galileo seems content here, albeit MatekF405, BN880 and almost 2.2.1 (but we haven't touched the GPS code afaik).

DzikuVx, 2019-07-16:
Tested on 2.2.1 and I can not reproduce. On the other side, Galileo is now down and unknown when it will go up again. I'm closing it, but if anything new will happen, please let us know

closed 2019-07-16 by DzikuVx