iNavFlight/inav #4894: Display log capacity on OSD, with mode switch erasure of the logs

Jetrell, 2019-07-02:
1. I'd would like to request that the capacity of the log storage device, be displayed on the OSD. 
   So it can be known in flight.

2. Plus also have the ability to erase the log files in flight, via a mode switch selection.

 I think this would be helpful on long range flights. If the log storage device becomes full. And the user knows that nothing of interest was contained in those logs, but something of interest might take place later in the flight.

teckel12, 2019-07-03:
@Jetrell How long is this flight where you fill an SD card? ;) But seriously, are you always logging to SD card? I only do via a switch (technically a dial). I've been gun-shy to log all the time due to the occasional report about FC lockups due to SD card issues while trying to write. Maybe that issue has been corrected?

Jetrell, 2019-07-03:
@teckel12 Generally I only run old class 6, 8Gb cards. Nothing to expensive. I leave the good stuff for the HD.

I just let the log record continuously. Mainly because I forget what its capacity is up to. And looking at the capacity on the stats isn't always remembered.
That was the reason I also requested the ability to see the log capacity on the OSD during a flight. 
That would also help to know whether to enabled or disabled it via the mode.
Because I often leave it running. It may fill up over a days flying and I often forget about it. That then maybe the time when its needed.

I just figured that these two log features would be useful to have.

closed 2022-07-05 by Jetrell