iNavFlight/inav #4895: Maximum Rates in Manual Mode

MartinHugh, 2019-07-02:
I am using iNav2.1 with a 1m flying wing.

I am familiar with this airframe model with two different flight controllers (iNav and Vector) and without a flight controller at all.

I am noticing that with iNav the pitch and especially roll rates seem dumbed down.

Manual roll and pitch rates are set in iNav to 100%
Weights on all channels in the transmitter are set to 100%
There is no transmitter expo configured.

Is there something else I need to configure (like iNav expo?) or something else, to get to a real "manual" mode.  What else could be affecting the performance?

Thanks.

etulupov, 2019-07-02:
@MartinHugh 
Seems your config is ok. Please ensure that Manual Mode was activated.

You can also debug servo pulse values on Motor tab. You should see values from MIN to MAX range (typically from 1000 us to 2000 us).

MartinHugh, 2019-07-02:
@etulupov thank you for your response.  Yes, Manual Mode is activated.  I will check out the pulse values as you suggest.

b14ckyy, 2019-07-07:
check your mixer settings. By default it is at 50% maximum mix. Set this higher till you get the same throw like with a normal servo receiver. Be careful not to put too much throw in it to avoid highspeed stall but i think you are experienced enough to know that :)

MartinHugh, 2019-07-07:
@b14ckyy thanks for the heads up on that - I will check it out.

closed 2019-07-16 by DzikuVx