iNavFlight/inav #4899: RTH nav speed problems Matek 722-STD

neilm06, 2019-07-03:
## Current Behaviour
Quadcopter goes to set height but loiters when RTH engaged. Have to control speed manually. This behaviour is unpredictable as sometimes does not happen
 
## Steps to Reproduce

1. Attitude user control mode
2. Pos/Alt Hold engaged
3. Fly out suitable distance - 200m
4. Engage RTH

## Expected behaviour
When RTH engaged quadcopter should always go to set height and return at set max nav speed 600cm/s.
 
## Suggested solution(s)
Fix

## Additional context

---

- FC Board name and vendor: MATEK F722-STD
- INAV version string: INAV/MATEKF722 2.2.0 Jun  3 2019 / 14:05:06 (f59251378)

teckel12, 2019-07-03:
@neilm06 I'm running 2.2.0 on Matek F722SE and RTH works fine. If this is reproducible, please include FC config dump and SD log.

neilm06, 2019-07-04:
Thanks for getting back. I also have F722SE and no problems. This seems to be unpredictable on this F722-STD. The config dump is attached. The SD log is of the first flight of the evening. Went out about 1.5km began to get some VTX break and engaged RTH. OSD showed all ok - height decreased to that set at 25m but airspeed dropped to 0km/h (it was set to 600cm/s). Began to take over throttle and disengaged RTH. It was a bit hairy as VTX break up got worse as height dropped. Brought craft back to  around 200m and tried engaging RTH again. It engaged ok and craft came back correctly and landed. It then happened again on two further short flights. (Have tried to attach Log file but is too large- will try to copy, cut and paste relevant sections) 

 version
# INAV/MATEKF722 2.2.0 Jun  3 2019 / 14:05:06 (f59251378)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

# resources

# mixer
mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix

# servo

# logic

# feature
feature -TX_PROF_SEL
feature -TELEMETRY
feature GPS
feature BLACKBOX
feature PWM_OUTPUT_ENABLE

# beeper

# map

# serial
serial 0 64 115200 38400 0 115200
serial 1 1 9600 38400 0 115200
serial 3 2 115200 38400 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1325 2075
aux 1 1 0 1325 2075
aux 2 9 1 1300 1675
aux 3 8 1 1675 2075
aux 4 19 2 1300 1675

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout
osd_layout 0 0 13 12 V
osd_layout 0 1 18 14 V
osd_layout 0 3 8 6 H
osd_layout 0 4 8 6 H
osd_layout 0 7 12 12 H
osd_layout 0 9 1 2 H
osd_layout 0 11 8 14 V
osd_layout 0 12 1 4 H
osd_layout 0 13 23 12 V
osd_layout 0 14 1 3 H
osd_layout 0 15 3 12 V
osd_layout 0 20 19 14 H
osd_layout 0 21 19 15 H
osd_layout 0 22 17 1 V
osd_layout 0 23 11 1 V
osd_layout 0 28 23 11 H
osd_layout 0 30 1 8 V
osd_layout 0 33 23 9 H
osd_layout 0 40 21 13 H
osd_layout 0 53 12 14 H
osd_layout 1 13 23 1 V
osd_layout 1 30 0 0 V
osd_layout 1 40 1 2 V

# master
set looptime = 500
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 90
set gyro_stage2_lowpass_hz = 175
set acc_hardware = MPU6500
set acczero_x = 13
set acczero_y = -41
set acczero_z = -34
set accgain_x = 4085
set accgain_y = 4085
set accgain_z = 4082
set align_mag = CW270FLIP
set mag_hardware = HMC5883
set magzero_x = 87
set magzero_y = -57
set magzero_z = -34
set baro_hardware = BMP280
set pitot_hardware = NONE
set rssi_channel = 14
set serialrx_provider = IBUS
set rx_min_usec = 990
set blackbox_rate_num = 4
set blackbox_rate_denom = 5
set min_throttle = 1060
set max_throttle = 2000
set motor_pwm_rate = 4000
set motor_pwm_protocol = MULTISHOT
set failsafe_delay = 15
set failsafe_procedure = RTH
set align_board_roll = -26
set vbat_scale = 1087
set current_meter_scale = 256
set model_preview_type = 3
set 3d_deadband_low = 1425
set 3d_neutral = 1475
set gps_sbas_mode = AUTO
set gps_ublox_use_galileo = ON
set nav_wp_safe_distance = 12000
set nav_auto_speed = 500
set nav_manual_speed = 1667
set nav_manual_climb_rate = 300
set nav_rth_alt_mode = FIXED
set nav_rth_altitude = 2500
set nav_mc_bank_angle = 35
set nav_mc_hover_thr = 1310
set osd_rssi_alarm = 25
set osd_alt_alarm = 120
set osd_dist_alarm = 1300
set osd_current_alarm = 15
set name = Aether 8"

# profile
profile 1

set mc_p_pitch = 60
set mc_i_pitch = 40
set mc_d_pitch = 20
set mc_p_roll = 60
set mc_i_roll = 40
set mc_d_roll = 20
set mc_p_yaw = 80
set dterm_lpf_hz = 75
set tpa_rate = 10
set rc_yaw_expo = 70
set roll_rate = 60
set pitch_rate = 60
set yaw_rate = 50

# battery_profile
battery_profile 1

set vbat_max_cell_voltage = 410
set vbat_min_cell_voltage = 280
set vbat_warning_cell_voltage = 300

neilm06, 2019-07-04:
Cannot get the Log file smaller .... any advice please.
Meanwhile there seems to be strange behaviour with the stateFlag |SMALL_ANGLE|. I have written some observations in a Word doc.

[Matek F722 RTH problem.docx](https://github.com/iNavFlight/inav/files/3358834/Matek.F722.RTH.problem.docx)

neilm06, 2019-07-04:
[LOG00098.TXT.zip](https://github.com/iNavFlight/inav/files/3358941/LOG00098.TXT.zip)

Log as zip file .... still large

stronnag, 2019-07-04:
There is a consistent / persistent difference of opinion between the mag heading and GPS CoG. Most of the time this is around 20° (which may or may not be significant; most successful navigation MRs have much smaller differences (typically < 5°).  At some times, it much worse; see screenshot (this throws mwp's naive anomaly detector). The heading / CoG variation is worse as speed increases (not too bad below c. 10m/s, bad at 19/ms). 

The first (fail) RTH attempt the vehicle is moving at 14m/s directly away from home; the second is at 5m/s heading to home. In the former case, iNav will have to command much more power to slow / turn the craft, which may support a magnetic interference hypothesis
[image]
.

neilm06, 2019-07-04:
Thanks for the observation and work on the map plot. Little bit beyond me in regard to your reasoning for magnetic interference. I double checked the mag calibration with a Suunto compass and it is very nearly spot-on with N,E,S,W points. Though I don't know if this is relevant. Would your suggestion relate to version 2.2 firmware for this board?

Actually flashed the FC with Version 2.0.1 this afternoon - kept the same diff config. Did some runs with the craft and all was fine. RTH operated as it should every time and craft came back to less than a metre from home point. I have attached a log file of this.
[LOG00103.TXT.zip](https://github.com/iNavFlight/inav/files/3360377/LOG00103.TXT.zip)

version
# INAV/MATEKF722 2.0.1 Oct 31 2018 / 19:27:11 (a04e68aaa)

# resources

# mixer
mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix

# servo

# feature
feature -TX_PROF_SEL
feature -TELEMETRY
feature GPS
feature BLACKBOX
feature AIRMODE
feature PWM_OUTPUT_ENABLE

# beeper

# map

# serial
serial 0 64 115200 38400 0 115200
serial 1 1 9600 38400 0 115200
serial 3 2 115200 38400 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1325 2075
aux 1 1 0 1325 2075
aux 2 3 1 1300 1675
aux 3 9 1 1300 1675
aux 4 8 1 1675 2075
aux 5 19 2 1300 1675

# adjrange

# rxrange

# osd_layout
osd_layout 0 0 13 12 V
osd_layout 0 1 18 14 V
osd_layout 0 3 8 6 H
osd_layout 0 4 8 6 H
osd_layout 0 7 12 12 H
osd_layout 0 9 1 2 H
osd_layout 0 11 8 14 V
osd_layout 0 12 1 4 H
osd_layout 0 13 4 12 V
osd_layout 0 14 1 3 H
osd_layout 0 15 22 12 V
osd_layout 0 20 19 14 H
osd_layout 0 21 19 15 H
osd_layout 0 22 17 1 V
osd_layout 0 23 11 1 V
osd_layout 0 28 23 11 H
osd_layout 0 30 1 8 V
osd_layout 0 33 23 9 H
osd_layout 0 40 21 13 H
osd_layout 0 53 12 14 H
osd_layout 1 13 23 1 V
osd_layout 1 30 0 0 V
osd_layout 1 40 1 2 V

# master
set looptime = 500
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 90
set gyro_stage2_lowpass_hz = 175
set acc_hardware = MPU6500
set acczero_x = 13
set acczero_y = -41
set acczero_z = -34
set accgain_x = 4085
set accgain_y = 4085
set accgain_z = 4082
set align_mag = CW270FLIP
set mag_hardware = HMC5883
set magzero_x = 91
set magzero_y = -57
set magzero_z = -46
set baro_hardware = BMP280
set pitot_hardware = NONE
set rssi_channel = 14
set serialrx_provider = IBUS
set rx_min_usec = 990
set blackbox_rate_num = 4
set blackbox_rate_denom = 5
set min_throttle = 1060
set max_throttle = 2000
set motor_pwm_rate = 4000
set motor_pwm_protocol = MULTISHOT
set failsafe_delay = 15
set failsafe_procedure = RTH
set align_board_roll = -26
set vbat_scale = 1087
set current_meter_scale = 256
set model_preview_type = 3
set 3d_deadband_low = 1425
set 3d_neutral = 1475
set gps_sbas_mode = AUTO
set gps_ublox_use_galileo = ON
set nav_wp_safe_distance = 12000
set nav_auto_speed = 500
set nav_manual_speed = 1667
set nav_manual_climb_rate = 300
set nav_rth_alt_mode = FIXED
set nav_rth_altitude = 2500
set nav_mc_bank_angle = 35
set nav_mc_hover_thr = 1400
set osd_rssi_alarm = 25
set osd_alt_alarm = 120
set osd_dist_alarm = 1300
set name = Aether 8"

# profile
profile 1

set mc_p_pitch = 60
set mc_i_pitch = 40
set mc_d_pitch = 20
set mc_p_roll = 60
set mc_i_roll = 40
set mc_d_roll = 20
set mc_p_yaw = 80
set dterm_lpf_hz = 75
set tpa_rate = 10
set rc_yaw_expo = 70
set roll_rate = 60
set pitch_rate = 60
set yaw_rate = 50

# battery_profile
battery_profile 1

set vbat_max_cell_voltage = 410
set vbat_min_cell_voltage = 280
set vbat_warning_cell_voltage = 300

stronnag, 2019-07-04:
> Thanks for the observation and work on the map plot. Little bit beyond me in regard to your reasoning for magnetic interference.

Because it's the most common reason, and the symptoms match; mag and Cog show much better correlation when the current draw is < 5A compared to when it's > 5A. 

> I double checked the mag calibration with a Suunto compass and it is very nearly spot-on with N,E,S,W points. Though I don't know if this is relevant. 

Static verification doesn't really help with the problems caused by current related interference.

> Actually flashed the FC with Version 2.0.1 this afternoon - kept the same diff config. Did some runs with the craft and all was fine. RTH operated as it should every time and craft came back to less than a metre from home point. I have attached a log file of this.

Notable that the current draw doesn't exceed 5A when RTH is executed and during the return in this log, with good correlation. 

In the earlier phase of the flight, at higher current draw, there is significant difference between mag and CoG.

I'm not drawing any conclusions, just pointing out what a visualisation of the log shows.

stronnag, 2019-07-04:
https://youtu.be/nV9LLv8G-0o shows the replay of the 103 log. 
The mag / CoG correlates perfectly during the low draw RTH phase, less so during piloted flight.
The standout difference to me between 98 and 103 is the state (speed, draw) of the vehicle when RTH is initiated. But I'm just naively looking at the pictures ...

neilm06, 2019-07-05:
Thank you for the latest plot ..... and thank you for your work on MWP Tools. I never cease to be amazed at how clever people are. Any chance of running programme using Mac OSX?

To return to the issue: Indeed I have had instances of mag interference on other craft when GPS+Compass have been too close to sources. I those cases RTH has always kicked in but craft 'toilet bowled'. Have never had any problems with this particular craft until flashed with V2.2. Being an endurance craft the set up is low amp but have previously drawn 12amp on RTH with no problems - craft sometimes takes a bit of a dive due to my flying habits. I think your observations derived from MWP Tools is very helpful but I am still unsure why RTH fails to kick in with this craft since V2.2.

I am mindful not to load too many Log Files but have included one more which was the second incident of RTH problem on the same evening. It includes a failsafe.
[LOG00099.TXT.zip](https://github.com/iNavFlight/inav/files/3362738/LOG00099.TXT.zip)

stronnag, 2019-07-05:
Very strange behaviour in the '99 log. In particular the second RTH example where it starts, pauses, continues, stops going nowhere, at a pretty low draw. Baffled.

neilm06, 2019-07-08:
Wondered if any more thoughts on what this may be?

neilm06, 2019-07-13:
Have flashed 2.0.1 firmware and no problems. Think I will stay with this version until confident 2.2 firmware is fully stable on this board. As @stronnag has commented the behaviour is strange ... with no apparent cause confirmed.

neilm06, 2019-07-16:
Similar issue seems to show up on a Holybro Kakute F7 (non AIO) version 2.2.1. On different craft.

# version
# INAV/KAKUTEF7 2.2.1 Jul  3 2019 / 21:46:47 (a6d847482)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

# resources

# mixer
mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix

# servo

# logic

# feature
feature -TX_PROF_SEL
feature -TELEMETRY
feature GPS
feature PWM_OUTPUT_ENABLE

# beeper

# map

# serial
serial 0 1 9600 115200 0 115200
serial 2 2 115200 38400 0 115200

# aux
aux 0 0 0 1300 2050
aux 1 1 0 1300 2050
aux 2 9 1 1300 1625
aux 3 8 1 1625 2100
aux 4 19 2 1300 1625

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout
osd_layout 0 0 14 12 V
osd_layout 0 1 13 15 H
osd_layout 0 3 8 6 H
osd_layout 0 4 8 6 H
osd_layout 0 7 13 12 H
osd_layout 0 9 1 2 H
osd_layout 0 11 10 15 V
osd_layout 0 12 1 4 H
osd_layout 0 13 5 13 V
osd_layout 0 14 0 11 H
osd_layout 0 15 21 13 V
osd_layout 0 20 18 0 V
osd_layout 0 21 3 0 V
osd_layout 0 22 15 2 V
osd_layout 0 23 3 2 V
osd_layout 0 28 23 11 H
osd_layout 0 30 1 8 V
osd_layout 0 53 16 15 V

# master
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 70
set gyro_notch1_hz = 170
set gyro_notch1_cutoff = 125
set gyro_notch2_hz = 85
set gyro_notch2_cutoff = 43
set acc_hardware = ICM20689
set acczero_x = 13
set acczero_y = -24
set acczero_z = 2
set accgain_y = 4083
set accgain_z = 4030
set align_mag = CW270FLIP
set mag_hardware = HMC5883
set magzero_x = 80
set magzero_y = -34
set magzero_z = -162
set baro_hardware = BMP280
set pitot_hardware = NONE
set rssi_channel = 14
set serialrx_provider = IBUS
set rx_min_usec = 995
set blackbox_rate_num = 4
set blackbox_rate_denom = 5
set min_throttle = 1080
set max_throttle = 2000
set motor_pwm_rate = 2000
set motor_pwm_protocol = MULTISHOT
set failsafe_delay = 10
set failsafe_procedure = RTH
set vbat_scale = 1101
set current_meter_scale = 155
set current_meter_offset = 30
set bat_voltage_src = SAG_COMP
set model_preview_type = 3
set 3d_deadband_low = 1425
set 3d_neutral = 1475
set gps_sbas_mode = AUTO
set gps_ublox_use_galileo = ON
set nav_use_midthr_for_althold = ON
set nav_auto_speed = 400
set nav_rth_alt_mode = FIXED
set nav_rth_altitude = 2500
set nav_mc_hover_thr = 1355
set osd_alt_alarm = 120

# profile
profile 1

set mc_p_pitch = 80
set mc_d_pitch = 18
set mc_p_roll = 80
set mc_d_roll = 18
set mc_p_yaw = 95
set dterm_notch_hz = 125
set dterm_notch_cutoff = 90
set rate_accel_limit_roll_pitch = 3600

First flight: RTH no problems.   LOG00018.TXT.zip

Second flight: Problems with RTH. Flight went ok until RTH engaged (log 3min 29secs). Height drops ok – reaches fixed height – doesn’t move. After about 15secs I disengaged RTH then reengaged after a couple of seconds. RTH then proceeds to home position and lands.    LOG00019.TXT.zip

Third flight: Again there were problems with RTH (log 4min 36secs). Pretty nearly exact behaviour to flight 2.   LOG00020.TXT.zip

Observation on the logs: The amps were low (6-9amps) with all flights when RTH engaged. There was however a difference in heights when RTH engaged. The return  height was fixed in cli at 25m. First flight RTH engaged at 26m. Second flight engaged at 47m; third flight 42m.

Any thoughts on this?

neilm06, 2019-07-16:
[LOG00018.TXT.zip](https://github.com/iNavFlight/inav/files/3398048/LOG00018.TXT.zip)
[LOG00019.TXT.zip](https://github.com/iNavFlight/inav/files/3398050/LOG00019.TXT.zip)
[LOG00020.TXT.zip](https://github.com/iNavFlight/inav/files/3398054/LOG00020.TXT.zip)

neilm06, 2019-08-07:
Wonder if this is what is showing up here?
Multicopter RTH Behavior Changed In Version 2.2.1 #4994 as reported by Kenimhof 4th August

closed 2020-02-19 by DzikuVx