iNavFlight/inav #4902: Summary "min battery volt" is wrong

FinalFrag, 2019-07-05:
## Current Behavior
The minimum battery voltage in the disarm summary is higher than the actual minimum voltage during flight.
Video: https://www.youtube.com/watch?v=o0GWvwfioJY

## Steps to Reproduce
1. Fly too long ;)
2. Watch battery voltage during flight
3. Watch "min battery voltage" in disarm summary

## Expected behavior
min battery voltage 11.9 in the summary (hit at 0:02 in the clip). Or depending how often samples are taken during flight, it should at least report 12.2 as the battery is consistently at that voltage during landing.

## Suggested solution(s)
No suggestion

## Additional context
Notice that I am running "bat_voltage_src = SAG_COMP", but I also noticed it in 2.1 where I wasn't running SAG_COMP
Full diff: https://pastebin.com/MVp3R2Hi

---

- FC Board name and vendor: Matek F405 STD
- INAV version string: INAV/MATEKF405_SERVOS6 2.2.1 Jul  3 2019 / 22:32:52 (a6d847482)
GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

teckel12, 2019-07-06:
@FinalFrag Specify the cell count you fly with, all problems go away. Issue is with trying to calculate cell count with high voltage. Instead, just specify cell count and problems go away. Math...

FinalFrag, 2019-07-06:
Ok, I will retest

closed 2020-02-26 by DzikuVx