iNavFlight/inav #4903: Battery warning comes too late (blinking in OSD)

FinalFrag, 2019-07-05:
## Current Behavior
I have set up the battery warnings:
```
vbat_cell_detect_voltage = 450
vbat_max_cell_voltage = 420
vbat_min_cell_voltage = 330
vbat_warning_cell_voltage = 350
```

But I haven't set any capacity warnings:
```
battery_capacity = 0
battery_capacity_warning = 0
battery_capacity_critical = 0
```

## Steps to Reproduce
1. Fly
2. Get the blinking warning too late (at 12.0v)
3. Almost kill your battery

## Expected behavior
The battery voltage and capacity to start blinking at 14.0v (3.5v per cell)

## Suggested solution(s)
Fix? :)

## Additional context
Full diff: https://pastebin.com/MVp3R2Hi
Possibly related to https://github.com/iNavFlight/inav/issues/4902

---

- FC Board name and vendor: Matek F405 STD
- INAV version string:INAV/MATEKF405_SERVOS6 2.2.1 Jul  3 2019 / 22:32:52 (a6d847482)
GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

teckel12, 2019-07-06:
@FinalFrag The cell detect voltage seems too high, probably should be 4.3? Also, you can specify the cell count to avoid any problems.

FinalFrag, 2019-07-06:
Since the voltage starts blinking at 12.0v and the setting is at 3.5v, that means the system detected a 12/3,5 = 3,4 cell battery?

I'll try setting the cell count manually and test again, but it seems unlikely that the system would detect non-integer cell counts. I'll also add the cell voltage osd element in the next test.

teckel12, 2019-07-07:
@FinalFrag What was the charged voltage of your battery? 16.8v? 

Setting the cell count can just avoid any strange conditions, like installing a partially charged battery and it believes you have a different cell count.

FinalFrag, 2019-07-08:
The battery was plugged in at 16.6v, so that is without a doubt a 4S pack. I have set the cell count to 4 in the cli now, I will retest soon.

DzikuVx, 2020-05-27:
`vbat_cell_detect_voltage = 450` causes INAv to be not able to detect cell count and thus to compute cell voltage. Please revert this to the default value

closed 2020-05-27 by DzikuVx