iNavFlight/inav #4904: MATEK F405 WING 2.2.1 firmware does not support Blackbox logging.

SwiftFlying, 2019-07-06:
_2019-07-06 @ 10:47:37 -- MultiWii API version received - 2.3.0

2019-07-06 @ 10:47:37 -- Flight controller info, identifier: INAV, version: 2.2.1

2019-07-06 @ 10:47:37 -- Running firmware released on: Jul 3 2019 22:48:46

2019-07-06 @ 10:47:37 -- Board: MF4S, version: 0

2019-07-06 @ 10:47:37 -- Unique device ID received - 0x2b003b3236511832323539

2019-07-06 @ 10:47:37 -- PWM output is disabled. Motors and servos will not work. Use Configuration tab to enable!

2019-07-06 @ 10:48:22 -- Serial port successfully closed

2019-07-06 @ 10:48:39 -- USB device successfully opened with ID: 1

2019-07-06 @ 10:48:39 -- Detected device with total flash size 1024 kiB

2019-07-06 @ 10:48:39 -- Read protection not active

2019-07-06 @ 10:48:56 -- Erased 1024 kB of flash successfully

Blackbox
DOCUMENTATION FOR INAV
Your flight controller's firmware does not support Blackbox logging.

SwiftFlying, 2019-07-06:
[image]

FinalFrag, 2019-07-06:
Same issue on Matek F405 STD

```
# INAV/MATEKF405_SERVOS6 2.2.1 Jul  3 2019 / 22:32:52 (a6d847482)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]
```

FinalFrag, 2019-07-06:
Apparently on the configuration tab you have to enable "Blackbox flight data recorder" or use the cli ```feature BLACKBOX```

This fixed it for me on Matek F405 STD

closed 2019-07-06 by SwiftFlying

b14ckyy, 2019-07-07:
Why did you close this ticket? Even if we found a workaround this needs to stay open to be fixed. 
Same issue here on a freshly set up F405 Wing

FinalFrag, 2019-07-08:
@b14ckyy can you post your diff?