iNavFlight/inav #4910: Broker fixed_wing_auto_arm in version 2.2.1

tadamox, 2019-07-07:
## Current Behavior
CLI setting `fixed_wing_auto_arm = ON` does not allow to arm controller flashed with 2.2.0 or 2.2.1 version of INAV

## Steps to Reproduce

1. Get Matek F405 wing
2. Do a clean update to version 2.2.0 or 2.2.1
3. Configure basic airplane mixer
4. Enable motor and servo output
5. Set fixed_wing_auto_arm = ON
6. Noting happens after rising throttle

## Expected behavior
Motor starts spinning after rising throttle.
Version 2.1 works as expected

## Suggested solution(s)
Return behavior as in version 2.1

---

- FC Board name and vendor: Matek F405 Wing
- INAV version string: 2.2.0 and 2.2.1

hali9, 2019-07-09:
https://github.com/iNavFlight/inav/pull/4640#issuecomment-489354381 ?

digitalentity, 2019-07-09:
@hali9 well spotted. Fixed by #4922

closed 2019-07-09 by digitalentity