iNavFlight/inav #4911: OSD ladder and Artificial Horizon position in 2.2.x

V8Darren, 2019-07-08:
Since flashing INAV 2.2.x, the vertical ladder position and more importantly the AH level position has moved. In INAV 2.1 the vertical ladders were more central and the AH was dead centre when the model was level. The ladders you can live with, they just looked better positioned before. But the AH for level flight is off by quite a way. 

Here is an image showing the 2.1 and 2.2 ladder and AH level positions. Both are running the exact same settings using the diff from 2.1 and both have the appropriate versions default font loaded. I have added the yellow crosshairs to show the horizontal and vertical centres of the images.

[image]

As you can see, the AH in 2.1 is perfect, but 2.2 it is low.

- FC Board name and vendor: Matek F405-STD
- INAV version string: 2.1 & 2.2

V8Darren, 2019-07-08:
Just to confirm, this issue is in both 2.2.0 and 2.2.1

FinalFrag, 2019-07-10:
Just an idea, did you reflash the fonts when upgrading to 2.2.x? I know the fonts were changed to support the radar functionality.

V8Darren, 2019-07-10:
Sorry, I thought I put that in the bug report. With each version of INAV I uploaded the correct fonts to the FC. So 2.1.0 had the 2.1.0 fonts and 2.2.1 had the 2.2.1 fonts.

Arvycka, 2019-07-11:
Confirmed, the AHI is lower by one line in 2.2.1. There was a similar bug bug introduced in 2.1 RC2 where the AHI would sit one line too high and was related to PAL/NTSC detection. 
You can go to OSD settings, OSD-HUD-HORIZIN OFFSET and set it to -1 for now. But it still needs to be fixed I think.

OlivierC-FR, 2019-07-11:
It's just because it's computed a different way now, intead of being top left, it's bottom right, but anyway it will always be offset horizontally since there's an even numbers of columns (30). A workaround for this could be to spread the crosshair pixels over 2 or 4 chars instead of 3 right now, but then it would still be wrong vertically because PAL has 16 lines so same issue. Where NSTC has 13 (6+1+6, so vertical centering is possible). Having different crosshairs in the font, switched depending of PAL or NTSC, would be a waste of resources IMHO. I don't see any easy fix for this, until the truly graphical OSD arrive.

Arvycka, 2019-07-11:
Then with "HORIZON OFFSET" setting in the OSD menu the default AHI position doesn't really mater I suppose, as it can be set to one's preference.

V8Darren, 2019-07-14:
Oliver, it’s not the crosshair that’s really the issue. The ladders and crosshair moving can be adjusted by moving OSD data on the screen. 

However, the AHI always used to be level with the horizon and now is noticeably off. This is where the real problem lies. 

I understand that there are only a limited number of character spaces on the display. But if the location of the 0,0 coordinate was changed from top left to bottom right, why were the positions of the static elements (ladders, crosshair, etc) not adjusted by that 1 character so that they were in the same position? As I said, I don’t really care about those, but the level AHI should be as close to the centre of the display as possible. With 2.1 it was pretty damn close, with 2.2 it’s low. You can see this by the yellow centre lines on the posted images.

gmacfpv, 2019-08-03:
Yes I am having this issue since updating to 2.2.1. Both my crosshair and ladders have moved 1 column to the right and is now off center. 
[image]

gmacfpv, 2019-08-03:
Seeing as "HORIZON OFFSET" has been implemented, could we have a "VERTICAL OFFSET" also implemented?

OlivierC-FR, 2019-08-07:
There's an even number of rows, the real physical center is where I draw the yellow lines, so actually there's no "correct before" / "wrong now". Instead its "before = 1 char wrong to the left" / "now = 1 char wrong to the right" :)

[image]

Jetrell, 2019-08-15:
@OlivierC-FR I get your meaning. 
Although the main issue, is that the Configurator OSD preview, now does not line up with the OSD.
This is the reason that we are all noticing the offset.
I imagine that this will be continually mentioned because the OSD is not displaying what the setup preview says it should.
Would you be interested in fixing the configurator preview, so it works with the new AHI/Cross hair alignment?

Thanks

Jetrell, 2019-08-18:
@OlivierC-FR I understand what you mean about the odd number of vertical columns. So it isn't possible to have the AHI/Crosshair centered. 
But someone put a lot of thought into to this in the pasted. I have looked at this offset and measured it up; with multiply different monitors/goggles. And the offset is always closer to center when it is placed to the left, than now, when its to the right. (2-4mm depending on the monitor)
It maybe due to horizontal shift on most monitors/DVR's.

And as a side note. At present the OSD configurator preview doesn't line up with the OSD, because of this change.
If a symbol is placed directly above or below the AHI on the OSD preview. The AHI always appears offset to the right of that character.

closed 2019-12-24 by DzikuVx