iNavFlight/inav #4924: S5 output on MATEK 722-SE does not work in DSHOT

nathanjel, 2019-07-10:
## Current Behavior
In DSHOT mode (all speeds) S5 output on MATEK 722-SE does not operate correctly, there is no motor control, neither any ESC beeps to confirm correct startup. Tested with different ESC and cable connections to be sure. In STANDARD mode (PWM) all is OK.

## Steps to Reproduce
MATEK 722-SE flight controller flashed with most recent stable INAV (2.2.1)
I use 6 identical ESC's - Holybro Tekko32 F3 Metal 6S 65A
2019-07-10 @ 17:25:40 -- Flight controller info, identifier: INAV, version: 2.2.1
2019-07-10 @ 17:25:40 -- Running firmware released on: Jul 3 2019 22:44:40
2019-07-10 @ 17:25:40 -- Board: MF7S, version: 0
Setup any DSHOT mode

## Expected behavior
I would expect DSHOT control on all outputs S1-S6 on the controller.

## Suggested solution(s)
It seems this issue was somehow foreseen as if You check the sources, https://github.com/iNavFlight/inav/blob/2.2.1/src/main/target/MATEKF722SE/target.c on line 37 mentions something that S5 is in a clash with S2. I guess this could be addressed by different DMA settings, but I lack the STM32 experience to take the next step. But happy to follow a suggestion, test and make a pull request if I fix it.

## Additional context
Dump link: https://pastebin.com/54idtVh8
# version
# INAV/MATEKF722SE 2.2.1 Jul  3 2019 / 22:44:40 (a6d847482)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

digitalentity, 2019-07-10:
This is hardware limitation and there are no other option for DMA selection.

S2 - the only possibility is TIM3_CH2, DMA 1, stream 5, channel 5
S5 - the only possibility is TIM2_CH1, DMA 1, stream 5, channel 3

To make DSHOT work DMA/Stream should be unique for all channels, therefore there is no possibility to have more than 4 channels DSHOT on this FC.

closed 2019-07-10 by digitalentity

nathanjel, 2019-07-10:
Thank You. Is this due to how @MATEKSYS assigned hardware pins, or is this a more general STM32F7 limitation? Asking, as I wonder if hacking the 722-SE would be of any use.

digitalentity, 2019-07-11:
That's due to how manufacturers assign output pins. Generally, F722 has enough DMA channels to have more DSHOT outputs, but other design compromises might have to be made.